openhwfoundation/cva6 #1740: [BUG] : PMP CSRs

AyoubJalali, 2023-12-29:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
In the RISCV spec section **3.7 Physical Memory Protection** : 
[image]

So in the pmpxcfg has the combination of X, W, R is reserved when W = 1 & R = 0 (as X, W, R form a collective WARL), so if we write a value in pmpxcfg which have this combination, we should clear only the bits X, W, R (normally clear W), but the CVA6 clear the entire pmpxcfg.

ASintzoff, 2024-01-15:
Do you mean when writing `b'xxxxxx10` where x can be 0 or 1 in a `pmpxcfg` byte, the result is `pmpxcfg` equal to 0?

Moschn, 2024-04-16:
This issue is fixed and can be closed

AyoubJalali, 2024-04-16:
Yes you right, I double check the fix !!
Thanks @Moschn