openhwfoundation/cva6 #1748: uvme_cva6_cfg.sv uses hard constraints, prevents re-use by other CVA6 variants

MikeOpenHWGroup, 2024-01-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The `uvme_cva6_cfg_c` configuration class in [cva6/verif/env/uvme/uvme_cva6_cfg.sv](https://github.com/openhwgroup/cva6/blob/c430c6c34be0e24175e03d3c91478e349bfaaf4e/verif/env/uvme/uvme_cva6_cfg.sv#L29) has a large set of hard constraints that set the values of cfg members.  This means this cfg file is only usefull for one CVA6 variant, specifically the CV32A60X.

In order to make this configuration class reuseable, the hard constraints should either be removed (or set to "soft"), and the values of the members of `uvme_cva6_cfg_c` should be constrained in a variant-specific environment or testcase (the testcase is preferred).

This Issue is Labeled as both an "Enhancement" and a "Bug".  For now, as we are only working on the CV32E60X, making this change is an enhancment, as it will not change functionality for a single variant.  When we start supporting multiple variants of the CVA6, this Issue will become a bug.

MikeOpenHWGroup, 2024-01-05:
Hi @AyoubJalali, I am assigning this issue to you since you have made several contributions to `uvme_cva6_cfg.sv`.  Feel free to assign it to someone else if that is appropriate.

AyoubJalali, 2024-01-10:
Hello @MikeOpenHWGroup , This one of our next steps of enhace the UVM CVA6 env, we want to make this env re-usable using the RTL configuration (CVA6Cfg), in this way when the RTL change or the configuration of the RTL the CVA6 env should change automatically, the work is ongoing.

AyoubJalali, 2024-07-25:
@MikeOpenHWGroup could you check the uvme_cva6_cfg.sv, now it has values from the RTL configuration !

AyoubJalali, 2024-11-05:
Can we close this issue @MikeOpenHWGroup

JeanRochCoulon, 2025-01-02:
@MikeOpenHWGroup We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?