openhwfoundation/cva6 #1749: How high the main frequency can CVA6 run on FPGA?

chenqian13, 2024-01-06:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I increased the main frequency of CVA6 to 100MHz on the KU5P development board of Xilinx. Except for some garbled serial output when starting the Linux system(at the very beginning), everything else runs well. But when I increased the main frequency of CVA6 to 200MHz, there was a significant clock violation in the VIVADO project, and the Linux system could not start properly.

The paper mentions that the ASIC form of CVA6 can run up to a main frequency of 1.7GHz, but the performance of CVA6 on FPGA seems to have a significant gap compared to ASIC. Is this normal?

I want to know how high the main frequency can CVA6 run on FPGA?

JeanRochCoulon, 2024-01-07:
The max frequency uses to be higher on ASIC than FPGA, and that is the case with CVA6. @sjthales knows the max frequency we can reach on FPGA.

chenqian13, 2024-01-08:
@sjthales Hi! Could you please share your research on the maximum frequency that CVA6 can reach on FPGA？

demofpga, 2024-01-22:
@chenqian13 Like you, I upgraded the frequency to 100M'z, but vivado reports a lot of errors.  I used the fpga device is xilinx_artix_A100T(-2 speed grade). I modified the xlnx_clk_gen(CLK_OUT1=100MHz), and bootrom(for uart_init(100000000, 115200)。Can you tell me how you managed to do 100MHz, CVA6 work?  thanks for your help.

chenqian13, 2024-01-24:
@demofpga After I changed CLK_OUT1=100MHz, I updated cv64a6.dts under fpga/src/bootrom and regenerated bootrom_64.sv. Vivado did not report any errors. Additionally, I have updated cv64a6_genesysII.dts under cva6-sdk/uboot/arch/riscv/dts/ and regenerated the Linux image.

JeanRochCoulon, 2024-01-24:
Nice. @jquevremont do you think we need to update github projects to add this fix?

jquevremont, 2024-01-24:
Thanks for your interest in CVA6.
You can find some date about CVA6 performance on FPGA [here](https://riscv-europe.org/summit/2023/media/proceedings/posters/2023-06-07-J%C3%A9r%C3%B4me-QU%C3%89VREMONT-abstract.pdf). Contributions are welcome to extend the analysis of performance on FPGA.

demofpga, 2024-01-26:
> @demofpga After I changed CLK_OUT1=100MHz, I updated cv64a6.dts under fpga/src/bootrom and regenerated bootrom_64.sv. Vivado did not report any errors. Additionally, I have updated cv64a6_genesysII.dts under cva6-sdk/uboot/arch/riscv/dts/ and regenerated the Linux image.

@chenqian13 @jquevremont @JeanRochCoulon 
I also modified it in this way. The devices are different. My FPGA is Artix-A100T, not Kintex-K325. Maybe this is the reason for the frequency reporting error. What do you think?

Best Regards,

chenqian13, 2024-01-26:
@demofpga Can you show the details of the error reported by Vivado？

demofpga, 2024-01-28:
@chenqian13 
[cva6_2x_rpt.zip](https://github.com/openhwgroup/cva6/files/14074723/cva6_2x_rpt.zip)
I increased the main frequency of CVA6 to 100MHz on the xilinx_device artix-A100T(-2 speed grade). there was a significant clock violation in the VIVADO project, and question was forcus on the FPU. 

Please review my email attachments（timing report files, and constraint files).

Best,

closed 2024-03-04 by jquevremont