openhwfoundation/cva6 #1754: Remove deprecated CSR chapter from User Manual

JeanRochCoulon, 2024-01-10:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The User Manual includes the CSR file done a long time ago by Jade https://github.com/openhwgroup/cva6/blob/master/docs/01_cva6_user/CV32A6_Control_Status_Registers.rst. This created confusion in #1735. As this description is deprecated and wrong, can we remove it from this document, in the meantime we find a solution to specify the CSR whatever the configuration.

jquevremont, 2024-01-16:
First attempt in #1766

jquevremont, 2024-02-12:
@JeanRochCoulon, can you confirm the issue is closed?

JeanRochCoulon, 2024-02-12:
Indeed the PR is closed. Of course the CSR description need to be improved, this task is planned.

closed 2024-02-12 by JeanRochCoulon