openhwfoundation/cva6 #1759: [BUG] Undefined reference to functions specified in corev-apu/tb/ariane_tb.cpp file while running CVA6 on Verilator

Shreesh-Kulkarni, 2024-01-13:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I got this error while running CVA6 on verilator via the Makefile. 
[image]
After inspecting the `ariane_tb.cpp` file, there are the 3 instructions declared as `extern` functions, most probably referring to a different file which the makefile/.cpp file is unable to fetch.
[image]

Would appreciate help regarding this and any suggestions regarding the cause of the error and how one would fix it.

JeanRochCoulon, 2024-01-15:
Hello @Shreesh-Kulkarni   thanks for the report.
Can you confirm you simulate executing cva6.py ?

Shreesh-Kulkarni, 2024-01-16:
Thank you for replying, @JeanRochCoulon. I simulated and executed cva6.py and I received the same error. I am attaching the screenshots for the same.
[image]
I inspected the log file and the same error popped up as attached below :
[image]

MarioOpenHWGroup, 2024-01-16:
The problem here is probably that spike is not installed, which commands have you run before the `cd ./verif/sim` ?

MarioOpenHWGroup, 2024-01-17:
Which version of verilator are you using? Verilator headers should include this `svdpi.h`.

bekbeis, 2024-01-17:
I have Verilator 4.028 installed globally, and Verilator 5.018 installed locally.

The `svdpi.h` is indeed present in the local directory of Verilator. However, it is absent in the global directory.

Could it be that `verif/regress/install-spike.sh` refers to the global version of Verilator instead of the local one? If yes, can you please tell me how I can fix that?

MarioOpenHWGroup, 2024-01-17:
Yes it is what is happening. The setup-env should have set it correctly, can you unset the `VERILATOR_INSTALL_DIR` and source the `regress/sim/setup-env.sh` again?

bekbeis, 2024-01-17:
That helped, thank you so much!

Shreesh-Kulkarni, 2024-01-17:
@bekbeis Did you try running it with just verilator?

bekbeis, 2024-01-19:
@Shreesh-Kulkarni not really, I just followed the instructions from the readme file

JeanRochCoulon, 2024-01-24:
Did you try ?   If it works, please comment and close the github issue.

bekbeis, 2024-01-24:
I tried running cva6 with just Verilator, and it seems to work for me (but this was after fixing my issue with Spike)

closed 2024-01-24 by JeanRochCoulon

Shreesh-Kulkarni, 2024-01-26:
Sorry for responding late. But I'm still facing the issue even after installing Spike.
[image]
The below image I have attached clearly shows that Spike has been installed in the tools directory.
[image]

Shreesh-Kulkarni, 2024-01-26:
Sorry for responding late. The issue was with the toolchain I had already installed. I reconfigured it and it works fine now. Sorry for the trouble.