openhwfoundation/cva6 #1772: [BUG] OpenOCD connection failed

Siris-Limx, 2024-01-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I've transplanted CVA6 APU to a non-

JeanRochCoulon, 2024-01-23:
@jquevremont    Do you know somebody who can help ?

jquevremont, 2024-01-23:
@Siris-Limx, before any deep diving, have you looked at https://github.com/ThalesGroup/cva6-eclipse-demo?

Siris-Limx, 2024-01-23:
@jquevremont  Yes, I know the basic knowledge about baremetal debugging. The log from OpenOCD tells me that the CPU never raise `dmstatus.allhalted` high, thus giving the error `failed to halt`.

jquevremont, 2024-01-24:
We have only published such support on Genesys 2 board. Unfortunately, we do not have enough resources to address other settings. Contributions are welcome to extend the ecosystem.

closed 2024-01-24 by jquevremont

yadu-kr, 2024-02-06:
@Siris-Limx I have the same OpenOCD error. I ported CVA6 to the ZCU102 board, and then tried to debug the design via JTAG. Calling OpenOCD gives me the following error:
```
$ openocd -f ariane.cfg
Open On-Chip Debugger 0.12.0+dev-g61c6b08-dirty (2024-01-05-16:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : TAP riscv.cpu does not have valid IDCODE (idcode=0xfffffffe)
Info : [riscv.cpu] datacount=2 progbufsize=12
Error: [riscv.cpu] Unable to halt. dmcontrol=0x80000001, dmstatus=0x00000c82
Error: [riscv.cpu] Fatal: Hart 0 failed to halt during examine
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: [riscv.cpu] Unsupported DTM version: -1
Error: [riscv.cpu] Could not identify target type.
```
The verbose log shows the same errors as yours as well. Were you able to solve the problem? My design also has no SPI and Ethernet peripherals. I'm not sure if that's the cause of the issue.

Siris-Limx, 2024-02-06:
> @Siris-Limx I have the same OpenOCD error. 

@yadu-kr Hello! The problem I met was due to the bootrom.
To be exact, the default bootrom contain some code that the CPU could not recognize, such as the initialization of SPI.
This will cause the `unable to halt` error.

To identify the problem, I ran the post-synthesis simulation in Vivado and found out that the CPU will just stall at some position in the bootrom. I changed the bootrom to a simply infinite loop, and the OpenOCD could work well.

I hope my experience will help you. :)

yadu-kr, 2024-02-06:
@Siris-Limx Thank you for the quick response!

>  I changed the bootrom to a simply infinite loop, and the OpenOCD could work well.

I'm assuming you changed the bootrom inside the `fpga/src/bootrom` folder, which contains the zero stage bootloader (ZSBL). Would it be possible for you to share the changes you made or your new bootrom files? Would be really helpful.

Siris-Limx, 2024-02-08:
@yadu-kr Hello! Following code is my `bootrom.S` in `corev_apu/bootrom`.

```
.section .text.start, "ax", @progbits
.globl _start
_start:
# bootrom.sv need to be functional in 64 and 32 bits,
# li s0, DRAM_BASE creates instructions not compatible with both
# versions. That's why we have replaced it by li and slli instructions
# to generates code compatible with both versions.
  li s0, 1
  slli s0, s0, 31
  jr _start
#  csrr a0, mhartid
#  la a1, _dtb
#  jr s0

.section .text.hang, "ax", @progbits
.globl _hang
_hang:
  csrr a0, mhartid
  la a1, _dtb
1:
  wfi
  j 1b

.section .rodata.dtb, "a", @progbits
.globl _dtb
.align 5, 0
_dtb:
.incbin "ariane.dtb"
```

Here, I just change to last sentence in `_start` section to `jr _start`, which means jump to the start of this section.
Obviously, this is an infinite loop.

To generate the corresponding `bootrom.sv`, you need to run `make` under `corev_apu/bootrom`, make sure you run `bash verif/sim/setup-env.sh` to set up the environment variables needed by the Makefile.