openhwfoundation/cva6 #1774: Simulation doesn't stop when running simple elf executables

yulong-lan, 2024-01-23:
I am using CVA6 to run a program with Verilator, the program is generated by a random instrucion generator.

A short example of assembly file (test.s) looks as below, and the reason why test.s has two sections relating to tohost and fromhost is that I get `warning: tohost and fromhost symbols not in ELF; can't communicate with target` if without them.
```
.section .text
.global _start

.section .tohost,"aw",@progbits
.align  6
.global tohost
tohost:  .dword 0

.section .fromhost,"aw",@progbits
.align  6
.global fromhost
fromhost:  .dword 0

_start:

    .word 0x73800093
    .word 0xca600113
    .word 0xff700193
    .word 0x56a00213
    .word 0xaea00293
    .word 0xbab00313
    .word 0x0b400393
    .word 0xfe300413
    .word 0xb4000493
    .word 0x39a00513
```

I use riscv gnu toolchain to compile it with command:
`riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -nostdlib -nostartfiles -o test test.s`

the disassembly looks as below, and it seems to be no problem:
```
test:     file format elf32-littleriscv

Disassembly of section .text:

00010054 <_start>:
   10054:       73800093                li      ra,1848
   10058:       ca600113                li      sp,-858
   1005c:       ff700193                li      gp,-9
   10060:       56a00213                li      tp,1386
   10064:       aea00293                li      t0,-1302
   10068:       bab00313                li      t1,-1109
   1006c:       0b400393                li      t2,180
   10070:       fe300413                li      s0,-29
   10074:       b4000493                li      s1,-1216
   10078:       39a00513                li      a0,922
```
However, when I run the simulation with this elf using Verilator and command:
`work-ver/Variane_testharness test`

Simulation simply keeps going forever, even when the number of instructions is very limited. Actually when I check the RVFI logfile, the processor is executing instructions that are completely different from the ones in the assembly file.

Can anybody help me as to point out at which step I am making a mistake?

yulong-lan, 2024-01-23:
Hello @JeanRochCoulon , thank you for your reply. 

Actually my branch is still the one from January 2023 instead of the latest, while I have made a lot of changes to the design and intend to keep them with that branch of CVA6. I guess it should not be affected by newer updates regarding the command. Do you think this is stil the issue?

JeanRochCoulon, 2024-01-24:
Thank for the information. We do not support too old hashes (and BTW deprecated commands) that's why I suggest to rebase (and use cva6.py) or to close the github issue.

closed 2024-02-10 by yulong-lan