openhwfoundation/cva6 #1778: When running the program, the output of the Linux terminal suddenly gets stuck.

chenqian13, 2024-01-24:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When I was running my own C++program, the output printed through the printf() function suddenly froze on the serial terminal. And I cannot make any input through the keyboard in the terminal. But through Vivado's Logic Analyzer, it can be seen that my program has actually completed the entire operation normally. I have tried different terminal tools and the phenomenon is the same. So I think it may be a hardware issue.

The output of the serial terminal gets stuck in the area shown in the picture, sometimes also getting stuck in "wrbuf11" and "wrbuf13". At that location, I configured some custom registers through the AXI bus. Did this operation affect the serial part? Has anyone encountered a similar situation as me?

Any help or clarification would be much appreciated. Thanks!

[image]
[image]

JeanRochCoulon, 2024-01-24:
@jquevremont     I labeled this PR as "Component:SW" but I do not know who to assign. I set your name, sorry for the bad gift. We should discuss it during cva6 meeting.

jquevremont, 2024-01-24:
@chenqian13, are you using https://github.com/openhwgroup/meta-cva6-yocto on Genesys 2 board?

chenqian13, 2024-01-24:
Hi! @jquevremont 
No, I'm using https://github.com/openhwgroup/cva6-sdk on xcku5p board. 
I added an axi master and axi slave in the design. I added the axi-slave device information (as shown in the figure) and changed the CPU clock-frequency and UART clock-frequency (both equal to CLK_OUT1=100MHZ) in "cv64a6_genesysII.dts" under "cva6-sdk/uboot/arch/riscv/dts/", and then remade the Linux system. In addition, I did the same thing for "cv64a6. dts" under "FPGA/src/bootrom" and regenerated the bootrom_ 64.sv.
Should I add my master device information in the dts files?

What's the difference between https://github.com/openhwgroup/cva6-sdk and https://github.com/openhwgroup/meta-cva6-yocto ? Should I use https://github.com/openhwgroup/meta-cva6-yocto to make Linux？ 

[image]

jquevremont, 2024-01-24:
cva6-dsk is based on BuildRoot. meta-cva6-yocto is based on Yocto.
May I suggest that you start your Linux system on a Genesys 2 board to get it up and running and help with your learning curve?

chenqian13, 2024-01-25:
@jquevremont OK, thank you for your quick reply！

I would greatly appreciate it if anyone else could continue to offer help or suggestions!

chenqian13, 2024-01-28:
Hi! @jquevremont 
I followed the README in the "meta-cva6-yocto" repository and finally generated the content shown in the figure.
Then I executed the following command: gunzip -c build/tmp-glibc/deploy/images/cv32a6-genesys2/core-image-minimal-cv32a6-genesys2.wic.gz | sudo dd of=/dev/sd$ bs=1M iflag=fullblock oflag=direct conv=fsync status=progress
But I was unable to successfully start the system. 
I want to know what the partition requirements for the SD card are. Also, I would like to know if there are any other files that need to be placed in the SD card besides the files in the command above.

[image]