openhwfoundation/cva6 #1783: How to get the `printf` output and .vcd file with new cva6.py

ytliu74, 2024-01-24:
Hi, I am new to CVA6 and try to run the standalone simulations according to [the instructions](https://github.com/openhwgroup/cva6?tab=readme-ov-file#running-standalone-simulations). It runs successfully, but I don't know where to check the output of `printf` function in my terminal. And I am also wondering how to generate the waveform (.vcd) files?

Thanks.

JeanRochCoulon, 2024-01-25:
@zchamski   I am sure you have the right answer.

zchamski, 2024-02-09:
Hi @ytliu74 , the instructions for generating waveform files (VCD, FST, VPD and FSDB) are now available in  https://github.com/openhwgroup/cva6/issues/1811#issuecomment-1934196656.  Can you please try them and report on the outcome?

These instructions were missing in the top-level README file of CVA6 - a fix for this is coming soon.

Regarding the availability and use of `printf` and related I/O functions: CVA6 activities are currently focused on verification, with environment interactions getting a low priority.  If I/O features are important for you, we could envision joint and coordinated actions to set up the necessary support.

zchamski, 2024-02-15:
Hi @ytliu74, can you please confirm that the VCD instructions added to issue #1811 and PR #1827 work for you?

If so, I propose that you close the current issue and create a separate issue for the `printf` topic (see also issue #1425).

ytliu74, 2024-02-16:
@zchamski Thanks so much. Yes, the VCD generation works.
And no problem, I'll close thie issue.

closed 2024-02-16 by ytliu74