openhwfoundation/cva6 #1785: [BUG] Allow one outstanding killed miss

JeanRochCoulon, 2024-01-25:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

@colluca has sublitted a PR to fix the following performance bug #1497. As the PR has not been merged. This github issue backup the problem.

The I$ can receive a speculative request from the instruction fetch unit (e.g. upon a branch). If this instruction is not in the cache we will incur a miss and a refill request will be issued to the memory. If, when the branch is resolved, the speculated instruction is killed, the refill request will be killed (i.e. the speculated instruction will never be loaded into the cache). Despite being killed, the I$ still waits on the request from memory to be back, stalling all other operation. This implies that even with a hot cache, we can still experience cache miss stalls on speculated instructions which will never enter the cache, and hence the cache will never truely be "hot" (in this sense, speculated instructions should be considered part of the working set).

This PR attempts to improve performance in this condition by handling up to one killed refill request asynchronously. A speculated miss will only induce a stall if another miss occurs before the killed refill returns from memory.

JeanRochCoulon, 2024-02-08:
Miek, I cannot assign @colluca to this issue, can you include him to the  CVA6 project ?

JeanRochCoulon, 2024-06-20:
Hello @yanicasa 
During your OBI investigation, you likely face this issue. Do you think it is solved in feature/interconnect branch ?

yanicasa, 2024-06-20:
In the current state of feature/interconnect, if the frontend kills a request when the icache's FSM is in MISS, the frontend is freed directly (OBI valid), so it will be able to make a new request, but this request will be blocked as the MISS is not finished because the icache doesn't support HIT in a MISS (due to icache implementation). 

what has changed is that **non-idempotent speculative accesses no longer go to the icache**, they wait to be resolved in the frontend, so they **no longer create a MISS**.

In my opinion, @colluca's PR is always interesting.

yanicasa, 2025-01-24:
I think this issue is only related to the way icache manage refill mechanism. 
As CV32A65X is PipelineOnly, label can be updated.