openhwfoundation/cva6 #1791: [TASK] HPDcache evaluation

JeanRochCoulon, 2024-01-29:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

1) Make HPDcache synthesisable by modifying the CVA6 synthesis scripts.
2) Modify the HPDCache input paramters to configure HPDCache, the objective is to fit embedded applications: low gate count and low memory size. The aaim is to be as small as write through cache.

### Required Changes

1) Modify the synthesis scripts to be able to read the HPDCache RTL
2) Modify the RTL HPDCache input parameters and condition unwanted RTL blocks in RTL. Gate count should reduce, Code coverage should increase

### Current Status

Today,
1) Synthesis script does not run when HPDcache is selected
2) Too much HPDcache functionalities are implemented in HPDCache

### Risks

Design flow need to be understood to be able to run a synthesis, run a simulation, produce a FPGA bit stream

### Prerequisites

Write through cache KPIs are the reference, they should be produced to be compared with HPDCache KPIs
Data cache benchmark need to be identified

### KPI (KEY Performance Indicators)

Gate count and memory size
Code coverage
Benchmark performance

### Description of Done

When CC of HPDCache is 100% and when gate count, memory size are produced for HPDCache

JeanRochCoulon, 2024-05-19:
Hello @Gchauvon   Can you comment the task and close it if you think it need to be.

Gchauvon, 2024-05-20:
Evaluation is done. After some work on a small cache configuration, cv32a65x will have HPDCache.

closed 2024-05-20 by Gchauvon