openhwfoundation/cva6 #1803: Debugging CVA6 APU on a KC705 board

Juan-Gg, 2024-02-05:
I am attempting to get the CVA6 APU running on a KC705 board.

What I have done so far:

- Cloned the CVA6 repository, installed the tools and successfully ran smoke-tests.sh
- Ran $make fpga BOARD=kc705, and obtained the ariane_xilinx.mcs file.
- Programmed the kc705’s configuration memory using this file. I am using Vivado 2018.2.
- When pressing the “FPGA Prog” button, after about 3 minutes the “GPIO LEDS” start flashing, four in a counting pattern and the other four in a shifting pattern. No UART activity.
- I prepared an SD card with the Ariane 4.2 image (https://github.com/openhwgroup/cva6-sdk/releases), when powering on with this SD card I could not see any activity on the UART.
- Following the instructions here: https://github.com/openhwgroup/cva6?tab=readme-ov-file#debugging, I tried to debug the design. When I run “openocd -f ariane.cfg”, it does not find the core. I get the following output:

```
ubuntu@ubuntu-2204:~/Documents$ openocd -f ariane.cfg
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: Unsupported DTM version: 15
```
Any ideas on what may be going wrong? What do the LEDs mean? I could not find any documentation on these.
Thanks.

jquevremont, 2024-02-07:
The SDK is adapted to the Genesys 2 board, which has been selected by OpenHW Group for most of its projects.
As of now, the CVA6 team has no resource to provide such support for other boards. OpenHW members are welcome to propose such new projects and contributions.

jquevremont, 2024-02-07:
I need to update the README to state the limitation.

Juan-Gg, 2024-02-07:
But even without sdk support, shouldn't I be able to at least see the core on the JTAG chain, with OpenOCD?

On the README for this repository, it does say:
> We currently only provide support for the Genesys 2 board.

But the makefile has support for kc705 & vc707 boards, related scripts for these boards are in corev_apu/fpga/src and corev_apu/fpga/constraints. I understand these are not maintained?

jquevremont, 2024-02-07:
Your understanding is right. More contributors to OpenHW projects are welcome to add more such maintenance.

Juan-Gg, 2024-03-18:
Here is what I did to get it working. Hope somebody finds it useful.

### Generating the bitstream
The LEDs I was mentioning have nothing to do with CVA6. There is no GPIO implemented at the time of writing.
Programming the board with the MCS file did not work for some reason, but using the bitstream directly did work and the board booted. This is what the bootrom outputs, if the SD card with the linux image is present, it will boot Linux next.

```
..Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
initializing SD... 
```
In order to look at the Vivado project and be able to see the elaboration results, I changed the target in corev_apu/fpga/Makefile from \$\(mcs) to \$\(bit). This way the Vivado project is not overwritten for MCS generation.

**Setting up the JTAG with BSCANE2**
Turns out the Genesys 2 FT2232 IC has two JTAG ports broken out, one for the FPGA, one free to use, in this case, for debugging the cva6.
The kc705 board, even though having the same IC, only has one JTAG port, used for the FPGA. The cva6 JTAG pins are connected to the FMC connectors, and could be used with an external JTAG adapter.
Another option is using Xilinx BSCANE2 IPs, which expose some additional register in the FPGA’s JTAG, which can be used to debug the processor through the FPGA’s JTAG.
This functionality is available as part of the [riscv-dbg repository ](https://github.com/pulp-platform/riscv-dbg), though a newer version than what is used currently in cva6.

The file “dmi_bscane_tap.sv” can replace “dmi_jtag_tap.sv” to use Xilinx BSCANE2 IPs instead of external pins. As they note “The file is pin-compatible so that by selecting the appropriate file for the target it can be transparently managed without relying on tick defines.”
Update the riscv-dbg submodule to get the “dmi_bscane_tap.sv”.
To synthesize correctly it needs an updated version of the common_cells submodule.
Replace “dmi_jtag_tap.sv” by “dmi_bscane_tap.sv” on the Makefile.
Massage the included sources until it synthesizes (after this shenanigans the Verilator simulation no longer works).

Note: It may be a good idea to update riscv-dbg and common_cells to more recent versions so this BSCAN option is available by default.

### OpenOCD configuration
Then, one can launch OpenOCD ([setup](https://github.com/openhwgroup/cva6?tab=readme-ov-file#debugging), [download](https://github.com/xpack-dev-tools/openocd-xpack/releases/), [install](https://xpack.github.io/dev-tools/openocd/install/))

I got it to work with the following configuration script, adapted from [here](https://github.com/esl-epfl/x-heep).

```
# Adapted from core-v-mini-mcu-pynq-z2-bscan.cfg

adapter driver ftdi     
adapter speed 1000
transport select jtag

# FT2232HQ Adapter
ftdi_vid_pid 0x0403 0x6010

ftdi_channel 0
ftdi_layout_init 0x0088 0x008b

reset_config none

echo "ftdi setting..."

set _CHIPNAME riscv

jtag newtap $_CHIPNAME cpu -irlen 6 -expected-id 0x43651093

set _TARGETNAME $_CHIPNAME.cpu

target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0x000

echo "target created..."

#log_output openocd_fpga.log

riscv set_ir idcode 0x09
riscv set_ir dtmcs 0x22
riscv set_ir dmi 0x23

riscv set_reset_timeout_sec 2000
riscv set_command_timeout_sec 2000

echo "setting preferences..."

scan_chain

init

echo "init routine started"

halt

echo "Ready for Remote Connections"

```
### Running OpenOCD:
A sample execution:
```
$ openocd -f bscan_modified.cfg
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:03)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
ftdi setting...
target created...
setting preferences...
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000014112f
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
init routine started
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

```
### Running GDB (This onwards applies to any board):

elialaz, 2024-03-29:
Hi juan,

Could you plis share a baremetal c project example? 
I'm quite stuck and I cannot run anything successfully unfortunately.

Thanks

Juan-Gg, 2024-03-31:
@elialaz, I have added a link on my previous post. I hope it works for you.

elialaz, 2024-03-31:
Thanks very much!

isa084, 2025-02-22:
Bookmarking this comment as it was helpful for running the baremetal example on Genesys 2 as well. I was able to fully reproduce @Juan-Gg 's "Hello CVA6" print