openhwfoundation/cva6 #1810: [TASK] AHB peripheral bus integration

CoralieAllioux, 2024-02-06:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Integration of a dedicated bus to program external peripherals. It avoids to go through AXI bus in that usecase.

### Required Changes

- Changes for data scratchpad (#1808)
- Changes for instruction scratchpad (#1809)
- Add AHB master interface to cva6 top
- Add controller to interact with the core and AHB interface
- Memory map update: dedicated space for external peripherals

### Current Status

RTL done, verification in progress

### Risks

cva6 interface update and memory map update.
Could have an impact on max frequency if enable, depending on the instruction scratchpad size.

### Description of Done

PR done in cva6 master. AHB peripheral bus functional for dedicated configuration, with dedicated tests.

JeanRochCoulon, 2024-02-07:
Could yuo give more information about the AHB property: AHB version, burst, virtualization,... ?

JeanRochCoulon, 2024-02-08:
Thank you for the AHB information. I assigned the tasks (this one and others relative to scratchpads and MMU) in relation to what has been discussed during the workshop. You can re-assign if needed.