openhwfoundation/cva6 #1811: how to get the .vcd file from the simulations

tushar-masur, 2024-02-06:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hey guys, i am new to this stuff (linux and github) here, 
so i basically followed all the steps mentioned in the [quick start](https://github.com/openhwgroup/cva6?tab=readme-ov-file#quick-setup) and running standalone simulations and managed to get the output, but i dont know how to get the .vcd file from these simulations using verilator, can anyone help me regarding this...

PS: i know this is not a bug, but i was not sure where to post questions...

JeanRochCoulon, 2024-02-07:
Thank you for your github issue. I confirm, the VCD generation is down. To be fixed.

tushar-masur, 2024-02-07:
> Thank you for your github issue. I confirm, the VCD generation is down. To be fixed.

Ohh Thankyou..

JeanRochCoulon, 2024-02-07:
Keeping the issue open allows to track the issue before fixing it.

tushar-masur, 2024-02-07:
> Keeping the issue open allows to track the issue before fixing it.

sorry, my bad.

tushar-masur, 2024-02-09:
> Hi @tushar-masur, waveform generation is disabled by default to save simulation time and storage space. It can be enabled by setting one of the two shell variables that control tracing, prior to running any of the test scripts under `verif/regress`:
> 
> * `export TRACE_FAST=1` enables "fast" waveform generation (keep simulation time low at the expense of space).  This will produce VCD files when using Verilator, and VPD files when using Synopsys VCS.
> * `export TRACE_COMPACT=1` enables "compact" waveform generation (keep waveform files smaller at the expense of increased simulation time).  This will produce FST files when using Verilator, and FSDB files when using Synopsys VCS.
> 
> Obviously, the two settings above are mutually exclusive and the scripts will throw an error if both variables are defined at the same time.
> 
> After each Verilog/VCS simulation the waveforms will be copied to the directory containing the log files, with the current HW target config name added to the file name right before the file type suffix (e.g., `.cv32a60x.vcd`).
> 
> Here's an example (tested with the latest CVA6 master commit `3f8649ec`):
> 
> ```
> export DV_SIMULATORS=veri-testharness,spike
> export TRACE_FAST=1
> export RISCV=.....your_gcc_installation_dir
> bash verif/regress/smoke-tests.sh
> ```
> 
> After the test script completes, the `verif/sim/out_<DATE-OF-TODAY>/veri-testharness_sim` directory will contain a set of files with `.vcd` extension.
> 
> Clearly, the above information is missing in the README.md file, so it needs to be added there (either directly, or by reference).

Thankyou

zchamski, 2024-02-09:
Hi @tushar-masur, from the two 👍 I guess things worked out OK 😃 Can you please confirm?

tushar-masur, 2024-02-09:
> Hi @tushar-masur, from the two 👍 I guess things worked out OK 😃 Can you please confirm?

Yes it worked😀, btw can you tell how to simulate custom programs on it

zchamski, 2024-02-09:
@tushar-masur, I propose that you open a new CVA6 issue on "How to add custom tests to the CVA6 environment?".  The knowledge is there, it just needs to be documented (pretty much like for the generation of waveforms).

In parallel I will add a new "Waveform generation" section to the CVA6 `README.md`, and once the corresponding PR is merged, I will ask you to close the current issue (#1811) as "completed".

zchamski, 2024-02-13:
Changes to documentation (and fixes to waveform generation under VCS) have been implemented in PR https://github.com/openhwgroup/cva6/pull/1827.

tushar-masur, 2024-02-13:
> Hi @tushar-masur, since PR #1827 has been merged, please close the current issue and create a new one about adding custom tests (see [#1811 (comment)](https://github.com/openhwgroup/cva6/issues/1811#issuecomment-1935965764) above).

Yes i have raised another issue regarding that in #1822 , Thankyou

closed 2024-02-13 by tushar-masur