openhwfoundation/cva6 #1835: Not able to get waveform using verilator (vcd file, fst file)

dvusingh, 2024-02-15:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hi,

I am trying to generate waveform in Verilator  for the median(benchmark test) in CVA6 but waveform(vcd file) is not generating after the simulation is completed successfully.

I tried the below options for DV_TARGET=cv64a6_imafdc_sv39:
export DV_SIMULATORS=veri-testharness,spike
export TRACE_FAST=1
bash verif/regress/benchmark.sh

After the test script completes, the verif/sim/out_<DATE-OF-TODAY>/veri-testharness_sim directory is created which contains only median.csv, median.log and median.log.iss files only. There is no file with .vcd extension.

PS: I was not sure where to post questions... for this issue, so I posted here

JeanRochCoulon, 2024-02-20:
@zchamski will help in understanding the issue. As first question, I would be pleased to know whether you pulled the last commits, because Zbigniew has fixed it last week.

dvusingh, 2024-02-21:
I pulled the last commits done by Zbigniew.
But still there is no vcd or fst file present in my veri-testharness_sim directory.

Thanks

zchamski, 2024-02-21:
Hi @dvusingh, please make sure that you activate Verilator-based RTL simulations and that you enable the waveform generation by setting one of the environment variables that control waveform generation:
* `export DV_SIMULATORS="spike,veri-testharness" ; export TRACE_FAST=1 ; bash verif/regress/smoke-tests.sh` should produce VCD files;
* `export DV_SIMULATORS="spike,veri-testharness" ; export TRACE_COMPACT=1 ; bash verif/regress/smoke-tests.sh` should produce FST files.

Since `TRACE_FAST` and `TRACE_COMPACT` variables are mutually exclusive, you may need to unset one of them to use the other.

zchamski, 2024-02-21:
Further discussion of waveform generation (VCS use, limitations etc.) is given in the toplevel `README.md` of the `cva6` tree (section "Waveform generation").

zchamski, 2024-02-26:
Hi @dvusingh, did you get a chance to test the instructions above?  If so, did they work for you?

dvusingh, 2024-02-26:
Hi @zchamski, I tried export TRACE_FAST=1 option and ran the smoke_tests.sh script. But still, I am not able to get the .vcd file in my log directory.

Note: I have installed verilator-v5.018 for cva6 verification

JeanRochCoulon, 2024-02-26:
For information, the supported Verilator is 5.005 (and not 5.018). I suggest to test with the supported version.

zchamski, 2024-02-27:
@JeanRochCoulon, did you mean 5.008?  Odd minor versions of Verilator are development/unstable ones.

dvusingh, 2024-03-01:
Hi @zchamski and @JeanRochCoulon, I installed the verilator-5.008 and tried running  smoke_test, but the job is terminated.

**g++: fatal error: Killed signal terminated program cc1plus**
compilation terminated.
make[2]: *** [/home/cva6/tools/verilator-v5.005/share/verilator/include/verilated.mk:275: verilated.o] Error 1
make[2]: *** Waiting for unfinished jobs....
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.

I also tried with DV_SIMULATORS=vcs-uvm, for this also I am getting the logs without any errors but the there is no vpd file created in log directory. 

Any suggestions?

dvusingh, 2024-03-11:
Issue resolved.

Thanks

closed 2024-03-11 by dvusingh