openhwfoundation/cva6 #1840: [BUG] Small typo in GCC toolchain builder "Getting Started" instructions

TommyMurphyTM1234, 2024-02-16:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The GCC toolchain builder "Getting Started" instructions:

* https://github.com/openhwgroup/cva6/tree/master/util/gcc-toolchain-builder#getting-started

include this text:
```
# 3. Build and install the toolchain (requires write+create permissions for $INSTALL_DIR.)
You can set 
sh build-toolchain.sh $INSTALL_DIR
```
It looks like "You can set" is prematurely truncated and probably should be a comment rather than a command?

valentinThomazic, 2024-02-22:
Hey @TommyMurphyTM1234, you are right ! Don't hesitate to submit a PR to fix the issue :)

valentinThomazic, 2024-02-22:
You can do what you deem the most appropriate, I would delete it.

JeanRochCoulon, 2024-03-07:
Done by #1901

closed 2024-03-07 by JeanRochCoulon