openhwfoundation/cva6 #1844: [BUG] mcountinhibit CSR

AyoubJalali, 2024-02-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

As the CV32A60X does not support _**mcountinhibit**_ , that's means we should only read-zero as the RISCV spec mentioned

ASintzoff, 2024-02-21:
> As the CV32A60X does not support _**mcountinhibit**_ , that's means we should only read-zero as the RISCV spec mentioned

The spec does not say that we should read zero. As the CSR is not implemented, there will be exceptions on read/write accesses.

AyoubJalali, 2024-02-21:
> The spec does not say that we should read zero. As the CSR is not implemented, there will be exceptions on read/write accesses.

could you see the section talking about **mcountinhibit**, in the RISCV spec volume 2 to be should, this is not how I understand it