openhwfoundation/cva6 #1871: How can I use CVA6 to run User-Space applications(e.g hello.c)?

niuzhi1221, 2024-02-24:
I'm new to use CVA6.Following  **setup** tutorial, I still can't find the way to run user-space applications.
I write a hello.c file, the content is as following:
#include<stdio.h>
int main(){
   printf("hello, cva6!\n");
   return 0;
}
I would appreciate if u could help me！Thanks！！

Juan-Gg, 2024-02-25:
See the instructions on the project readme [Running standalone simulations](https://github.com/openhwgroup/cva6?tab=readme-ov-file#running-standalone-simulations)

The path to your C file is specified when calling cva6.py, after --c_tests.

```
python3 cva6.py --target cv32a60x --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml \
--c_tests ../tests/custom/hello_world/hello_world.c \ <--####### PATH TO C FILE ########
--linker=../tests/custom/common/test.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles -g ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"
```
But you will not see any output from printf. See #748, #1783.
You can generate a .vcd waveform file though, by doing `export TRACE_FAST=1` before running the simulation. I found I needed to run `make clean && make -C verif/sim clean_all` for this to work.

JeanRochCoulon, 2024-02-25:
Thanks @Juan-Gg for the support.
@niuzhi1221 If this answer your question, please close the issue.

closed 2024-02-26 by niuzhi1221