openhwfoundation/cva6 #1872: Error in syscall.c while running hello_world test

KinzaQamar, 2024-02-24:
Hi,

I am getting an error inside **printf** function of verif/tests/custom/common/syscall.c as 

> /home/kinzah/Documents/cva6/verif/sim/../tests/custom/common/syscalls.c:405:(.text+0xf64): undefined reference to `__riscv_flush_icache'
> collect2: error: ld returned 1 exit status

The command ran was : 
```
python3 cva6.py --target cv32a60x --iss=$DV_SIMULATORS --iss_yaml=cva6.yaml \
--c_tests ../tests/custom/hello_world/hello_world.c \
--linker=../tests/custom/common/test.ld \
--gcc_opts="-static -mcmodel=medany -fvisibility=hidden -nostdlib \
-nostartfiles -g ../tests/custom/common/syscalls.c \
../tests/custom/common/crt.S -lgcc \
-I../tests/custom/env -I../tests/custom/common"
```

JeanRochCoulon, 2024-02-25:
You command is smoke-test's one, that is great.
Coudl you provide the Spike and compiler versions you use ?  They should be aligned on README.md description

@zchamski @valentinThomazic

KinzaQamar, 2024-02-27:
GCC version: 13.2
Spike was installed through available scripts.

JeanRochCoulon, 2024-02-27:
Only @zchamski can help you ;-)

JeanRochCoulon, 2024-03-05:
This works on our side.
Maybe an idea: if you have updated the core-v-verif submodule, you must remove Spike to rebuild it.

JeanRochCoulon, 2024-03-05:
Re-installing from scratch should work.

JeanRochCoulon, 2025-01-02:
@KinzaQamar We are cleaning the "Github Issues". I assume this issue is closed. May I ask you to close it ?

closed 2025-01-02 by KinzaQamar