openhwfoundation/cva6 #1875: Work-ver/Variane_testharness failed when runing an elf file

niuzhi1221, 2024-02-26:
I write a helloworld.c file, and then generate a helloworld binary file. After that, I execute :
**work-ver/Variane_testharness verif/tests/custom/hello_world/hello_world**
but it failed , just like this:
[image]

valentinThomazic, 2024-02-26:
Hey @niuzhi1221, this is not the correct way to run simulation with the CVA6. 
Please follow the README instructions.

niuzhi1221, 2024-02-26:
> Hey @niuzhi1221, this is not the correct way to run simulations with the CVA6. Please follow the README instructions.

Ok, I know it. So I must use cva6.py to run simulation by configuring something?

valentinThomazic, 2024-02-26:
Well, there is already a hello_world example in the Quickstart so I suggest you follow the example.

niuzhi1221, 2024-02-27:
> Well, there is already a hello_world example in the Quickstart so I suggest you follow the example.

Thanks so much!

closed 2024-02-27 by niuzhi1221