openhwfoundation/cva6 #1878: How to use CVA6.py to simulate a cpp file?

niuzhi1221, 2024-02-27:
I'm new to CVA6, but I can't use CVA6.py to run a cpp file（by using GCC Toolchain built following the Quick setup).Can someone provide me with certain instructions to help?I would be very grateful if you could provide me with your help.

Sincerely

valentinThomazic, 2024-02-27:
While it is true that `./verif/sim/cva6.py` does not directly support running cpp programs, you should be able to run .elf programs with it !

niuzhi1221, 2024-02-27:
> While it is true that `./verif/sim/cva6.py` does not directly support running cpp programs, you should be able to run .elf programs with it !

Thank you very much for your reply! So can I first use the riscv64 unknown elf gcc toolchain to compile Cpp files into. elf and. o files, and then use the run_elf in cva6. py to simulate? If it's like what I said, how should gcc_options be set? I installed it according to the requirements in Quickstart without riscv none elf g++. Thank you for your guidance.

Sincerely

valentinThomazic, 2024-02-27:
Yes exactly ! 
However, I am not sure I totally understood the last part of your message but the name of your toolchain tells me you did not install the toolchain from the repo scripts.
It is highly recommended to do so because it prevents a lot of versions/configurations error.

About the gcc _options I would try the same as the one showed in the README.

niuzhi1221, 2024-02-27:
> Yes exactly ! However, I am not sure I totally understood the last part of your message but the name of your toolchain tells me you did not install the toolchain from the repo scripts. It is highly [...]

Hello, I installed the toolchain according to repo scripts like this, there is no riscv-none-elf-g++:
[image]
How can I install the toolchain that can support C++ lanuage in scripts?

valentinThomazic, 2024-02-27:
Oh that's right, it does not by default.
You can set your own configuration for the toolchain build in `./util/riscv-toolchain/config` there is an argument which specify the supported languages when building the toolchain (you will need to build it again).

niuzhi1221, 2024-02-27:
I just find following configuration in `./util/gcc-toolchain-builder/config`:
[image]
So , you mean add "c++" in **--enable-languages** ? And then I rebuild it ?

valentinThomazic, 2024-02-27:
Yes did you do it ?

niuzhi1221, 2024-02-27:
I do it. But it failed.I'm so sad that I can't run elf_tests and cpp file.

valentinThomazic, 2024-02-27:
What is the issue?

niuzhi1221, 2024-02-27:
> What is the issue?

Can't install GCC successfully.Oh no

JeanRochCoulon, 2024-02-28:
This conversation is not related to CVA6 but to compiler tools. Please ask to GCC Github project for instance. Sorry but I need to close the github issue.

closed 2024-02-28 by JeanRochCoulon