openhwfoundation/cva6 #1880: RVFI : Missing CSRs in the RVFI implementation

AyoubJalali, 2024-02-29:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
Thanks to @yanicasa, now the RVFI implement CSRs in the design, but there's some missing CSRs.
The idea here is to implement all the RISCV supported CSRs, as we know the CVA6 has more then one configuration, so that's help us to filter the CSRs coming from the RVFI depend on the configuration.

it's a work should be done now or later

yanicasa, 2024-02-29:
Hello,
Most of the missing CSRs are related to performance counters. 
Do you have a particular missing CSR to add?

yanicasa, 2024-02-29:
do you have an exemple of registers name?

AyoubJalali, 2024-02-29:
> do you have an exemple of registers name?

like uie, ustatus, uip, ucause ...

JeanRochCoulon, 2024-03-11:
Any news on this issue ?

JeanRochCoulon, 2024-05-19:
@AyoubJalali Can you status on this Github issue?   Yannick and Côme have fixed CSRs several weeks ago, is it ok for you ?

AyoubJalali, 2024-05-20:
I think this is fine for me !!

closed 2024-05-20 by AyoubJalali