openhwfoundation/cva6 #1892: setup problem

michael116688, 2024-03-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

After I executed the commands:
export DV_SIMULATORS=veri-testharness,spike bash verif/regress/smoke-tests.sh
I encountered some problems.

> [no such file or directory ](ls: cannot access '/123/cva6/util/gcc-toolchain-builder/src/bin/riscv*-gcc': No such file or directory)

> Tue, 05 Mar 2024 23:24:08 ERROR ERROR return code: True/127, cmd:/123/cva6/util/gcc-toolchain-builder/src/bin/-gcc --version

[image]
[image]
[image]

JeanRochCoulon, 2024-03-05:
Have you installed the tools following the README ?     A env variable is not setup in /123/cva6/util/gcc-toolchain-builder/src/bin/-gcc

JeanRochCoulon, 2024-03-05:
BTW your issue is not a "BUG" but a "QUESTION" (so far)

michael116688, 2024-03-06:
> Have you installed the tools following the README ? A env variable is not setup in /123/cva6/util/gcc-toolchain-builder/src/bin/-gcc

I have reset my env variable, and installed the tools again. The problems were fixed.
Thank you very much.

closed 2024-03-06 by michael116688