openhwfoundation/cva6 #1894: Start Address points to 0x80000000

SShanmukh-cell, 2024-03-06:
The start address in the linker script (cva6/verif/tests/custom/common/test.ld) is always 0x80000000. 
I have tried to give address 0x00000000 to the start address of .text.init, but I am getting an error. 

So, is there any other files that I need to change, in order to make the code work with any start address?

JeanRochCoulon, 2024-03-06:
You changed the starting address in linker script, but CVA6 is configured to start at 80000000. You have to change both. I close the issue as it is resolved.

SShanmukh-cell, 2024-03-06:
Hi @JeanRochCoulon,
Can you please help me understand how to change the CVA6 start address or point out the file where the CVA6 is set to start at 0x80000000?