openhwfoundation/cva6 #1895: [Question] <Running regression tests simulations>

michael116688, 2024-03-06:
I run the riscv-arch-test regression test suite with the Verilator model by the following command.
```
export DV_SIMULATORS=veri-testharness,spike
bash verif/regress/dv-riscv-arch-test.sh
``` 
[image]

And, I get `add-01.cv64a6_imafdc_sv39.log.iss`. There are failed & error here. How could I fixed it?

[image]

JeanRochCoulon, 2024-03-07:
Please rebase your repository, and retry the command. A env checkup has been added Yesterday, it could help to debug.

JeanRochCoulon, 2024-03-11:
Does it work ?

michael116688, 2024-03-11:
> bash verif/regress/dv-riscv-arch-test.sh

It can start simulation, but when testing "B/src/zext.h-01.S" I met the error like bellow.
 
[image]

The .log.iss file is here.
[image]

JeanRochCoulon, 2024-03-11:
The simulation does not stop. I think we get the same behaviour on our side. @valentinThomazic or @cathales can you double check it ?

valentinThomazic, 2024-03-12:
Yes we do, the failed test is not caused by your environment which seems fine since you can run simulations.
As to why does the test timeouts, this is another question.

michael116688, 2024-03-12:
OK, thank you. 
I'll open another question for this.

closed 2024-03-12 by michael116688