openhwfoundation/cva6 #1897: Skip CI for documentation updates

jquevremont, 2024-03-06:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

To save processing power and licence usage, I suggest to skip the CI for pull requests where all files are in /docs.

cathales, 2024-03-08:
We currently [skip synthesis when `core/` and `corev/` are not modified](https://github.com/openhwgroup/cva6/blob/master/.gitlab-ci.yml#L192). However, GitLab does not read the diffs from the commits but [it compares with the "current"](https://docs.gitlab.com/ee/ci/yaml/#ruleschangescompare_to) version [of `master`](https://github.com/openhwgroup/cva6/blob/master/.gitlab-ci.yml#L196). This means that if `core/` changed on `master` in the meantime, it will be different from the version from the PR branch and synthesis will be run even if the PR does not change `core/`.  
Also note that `master` is updated on GitLab only twice a day to run regress tests.

RTL usually does not change a lot so we will likely not change this behavior.

About `docs/`, we could conditionally skip smoke (all) tests like in the same way, but it will still have the drawbacks described above.

If people working on docs *need* this feature now, I would delegate this task to @valentinThomazic.
Else I think this optimization should be discussed for potential later™ implementation.

jquevremont, 2024-03-08:
Hi @cathales,
This is not a need for people working on docs.
This suggestion is for a better usage of resources (licenses, computing farm, electricity). I feel embarrassed when I use them for a mere documentation update.
So I let you and your team decide to implement or ignore this request.
The side effet you describe is not an issue.

cathales, 2024-03-08:
I feel like it is not as easy as it seems.

* Not running a pipeline at all means no report, which will make us take the good habit to not wait for the report (which should be on all PRs)
* Having a successful report for no jobs run feels weird.

I don’t see a satisfying solution.

jquevremont, 2024-03-08:
Then let's close this.

closed 2024-03-08 by jquevremont