openhwfoundation/cva6 #1902: Condition coverage limitation

JeanRochCoulon, 2024-03-08:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

In the frontend.sv, the line 373 does not appear in the condition coverage report. To me, this is a condition coverage case. I am concerned because I think it hides an implementation bug. How to make it appear in the coverage report?

[image]

@AyoubJalali @ASintzoff @AEzzejjari

AyoubJalali, 2024-03-11:
Hello @JeanRochCoulon normally VCS can detect "?" condition by default, according to synopsys coverage manual :
[image]

also branch coverage don't detect this of course.
So what a thing we should re-wrote the line to be visible for condition coverage

JeanRochCoulon, 2024-03-11:
In the code, we have    **a = b ? c : d ;**    and   **not a <= b ? c : d ;**
That's maybe why it is not supported

AnouarZajni, 2024-03-15:
Checked some condition coverage options but can't make the coverage, I think because it's a limitation while mixing "+" operand and expression. 
To make it work we need to split to 2 lines

AnouarZajni, 2024-03-19:
Finaly, by splitting the line as following without any other condition coverage options the expression is covered:
[image]

AyoubJalali, 2024-03-19:
@AnouarZajni Great !!
But this is only an example, should we change in the hole design for that ? I don't think so

AnouarZajni, 2024-03-19:
Yes I can see many similar expression non-covered in design. You can find a non-exhaustive list in /shares/teams/riscv/fromAZ/missed_exp.txt
@JeanRochCoulon , can you please review the list?

JeanRochCoulon, 2025-01-02:
Sorry for the long delay, but this inpu tis very interesting. To be discussed.

AnouarZajni, 2025-01-08:
After some researchs, to enable such coverage we need to add "-cm_cond arith" in Makefile
@AyoubJalali will take care to update it. Waiting for the related PR

AnouarZajni, 2025-01-08:
Waiting for PR #2694 to close. 
Thanks @AyoubJalali

closed 2025-01-08 by JeanRochCoulon