openhwfoundation/cva6 #1912: [Question] Wrong control flow when execute an elf binary

yulong-lan, 2024-03-09:
I've written a random instruction generator to generate assembly files, which are compiled into elf using riscv64 gcc. 

the start address of the program starts at 0x80000000, in oder to make the processor only execute this program, I've set the `boot_addr_i` of the instance of `i_ariane` in `ariane_testharness.sv` (line 648) to 0x80000000 as well, presumably the processor should start fetching instructions from this memory location.

```
test_0:     file format elf64-littleriscv

Disassembly of section .text:

0000000080000000 <_start>:
    80000000:   61400093                li      ra,1556
    80000004:   00000013                nop
    80000008:   b1e00193                li      gp,-1250
    8000000c:   41a00213                li      tp,1050
    80000010:   9ae00293                li      t0,-1618
    80000014:   00000013                nop
    80000018:   00000013                nop
    8000001c:   00000013                nop
    80000020:   01700493                li      s1,23
    80000024:   00000013                nop
    80000028:   00000013                nop
    8000002c:   00000013                nop
    80000030:   efe00693                li      a3,-258
    80000034:   00000013                nop
    80000038:   e7000793                li      a5,-400
    8000003c:   23b00813                li      a6,571
    80000040:   8e900893                li      a7,-1815
    80000044:   3b000913                li      s2,944
    80000048:   00000013                nop
    8000004c:   00000013                nop
    80000050:   40600a93                li      s5,1030
    80000054:   7aa00b13                li      s6,1962
    80000058:   00000013                nop
    8000005c:   00000013                nop
    80000060:   00000013                nop
    80000064:   61f00d13                li      s10,1567
    80000068:   00000013                nop
    8000006c:   00000013                nop
    80000070:   ffc00e93                li      t4,-4
    80000074:   00200f13                li      t5,2
    80000078:   00000013                nop
    8000007c:   c7e48693                addi    a3,s1,-898
    80000080:   7b1b96a3                sh      a7,1965(s7)
    80000084:   0005eeb3                or      t4,a1,zero
    80000088:   e7d44083                lbu     ra,-387(s0)
    8000008c:   3e257f37                lui     t5,0x3e257
    80000090:   00addb33                srl     s6,s11,a0
```

Indeed it does, but then it repeatedly fetches instructions from 0x80000040 to 0x8000007c before it jumps to another random location of 0x800, And the rest of the executed instructions have nothing to do with my program. Apprently the control flow somehow changed many times even though there are no J or B type instructions at all, there are only` li `and `nop `instructions.

```
                 442 0x8000007c M (0xc7e48693) DASM(c7e48693)
                 453 0x80000040 M (0x8e900893) DASM(8e900893)
                 454 0x80000044 M (0x3b000913) DASM(3b000913)
                 455 0x80000048 M (0x00000013) DASM(00000013)
                 456 0x8000004c M (0x00000013) DASM(00000013)
                 457 0x80000050 M (0x40600a93) DASM(40600a93)
                 458 0x80000054 M (0x7aa00b13) DASM(7aa00b13)
                 459 0x80000058 M (0x00000013) DASM(00000013)
                 460 0x8000005c M (0x00000013) DASM(00000013)
                 461 0x80000060 M (0x00000013) DASM(00000013)
                 462 0x80000064 M (0x61f00d13) DASM(61f00d13)
                 463 0x80000068 M (0x00000013) DASM(00000013)
                 464 0x8000006c M (0x00000013) DASM(00000013)
                 465 0x80000070 M (0xffc00e93) DASM(ffc00e93)
                 466 0x80000074 M (0x00200f13) DASM(00200f13)
                 467 0x80000078 M (0x00000013) DASM(00000013)
                 468 0x8000007c M (0xc7e48693) DASM(c7e48693)
                 479 0x80000040 M (0x8e900893) DASM(8e900893)
                 480 0x80000044 M (0x3b000913) DASM(3b000913)
                 481 0x80000048 M (0x00000013) DASM(00000013)
                 482 0x8000004c M (0x00000013) DASM(00000013)
                 483 0x80000050 M (0x40600a93) DASM(40600a93)
                 484 0x80000054 M (0x7aa00b13) DASM(7aa00b13)
                 485 0x80000058 M (0x00000013) DASM(00000013)
                 486 0x8000005c M (0x00000013) DASM(00000013)
                 487 0x80000060 M (0x00000013) DASM(00000013)
                 488 0x80000064 M (0x61f00d13) DASM(61f00d13)
                 489 0x80000068 M (0x00000013) DASM(00000013)
                 490 0x8000006c M (0x00000013) DASM(00000013)
                 491 0x80000070 M (0xffc00e93) DASM(ffc00e93)
                 492 0x80000074 M (0x00200f13) DASM(00200f13)
                 493 0x80000078 M (0x00000013) DASM(00000013)
                 494 0x8000007c M (0xc7e48693) DASM(c7e48693)
                 514 0x800 D (0x00c0006f) DASM(00c0006f)
                 519 0x80c D (0x0ff0000f) DASM(0ff0000f)
                 529 0x810 D (0x7b241073) DASM(7b241073)
                 534 0x814 D (0x7b351073) DASM(7b351073)
                 539 0x818 D (0x00000517) DASM(00000517)
```

Could someone point me to the right direction?

JeanRochCoulon, 2024-03-10:
You get an exception (you jump into 0x800)
You execute "lbu     ra,-387(s0)"
Do you think this address value is valid?   That's maybe the exception cause.  You have to debug by analyzing the instruction trace or the waves.

yulong-lan, 2024-03-10:
Hi @JeanRochCoulon , thank you for your reply. According to the trace log above, I fail to understand why the processor repeatedly executes instructions between` li      a7,-1815` and `addi    a3,s1,-898`, where there are only `li `and `nop `instructions. It's never even gone so far as to `lbu     ra,-387(s0)`. This instruction is never executed.

JeanRochCoulon, 2024-03-10:
You cannot start executing instructions before configuring the CVA6. Stack pointers, register reset, etc... are done by CRT at the first begin of the execution. Please compile hello_world.c to obtain the needed instructions.

closed 2024-03-11 by JeanRochCoulon

yulong-lan, 2024-03-15:
Feedback:

That exception we see in the trace log starting at `514 0x800 D (0x00c0006f) DASM(00c0006f)` is not associated with instruction execution, i.e. it is not triggered by instruction. CVA6 makes use of RISCV Front-End Server (fesvr) to load program, in which a debug module is invoked periodically to access the memory, the result of which leads to this exception. The control flow will return to our program. If we don't want this to happen,  we need to disable this debug module feature.