openhwfoundation/cva6 #1918: [Question] <Running regression tests simulations: stop at test "B/src/zext.h-01.S">

michael116688, 2024-03-12:
It can start simulation, but when testing "B/src/zext.h-01.S" I met the error like bellow.
#1895 @valentinThomazic told that the test was timeout. How can I fix it?

[image]
The .log.iss file is here.
[image]

JeanRochCoulon, 2024-03-14:
When we face this kind of problem, we compare the log trace coming from Spike and RTL to understand the discrepancy, and try to solve the issue. Traces are under out_xxxx/

michael116688, 2024-03-19:
In this problem, I used the command 
`export DV_SIMULATORS=veri-testharness,spike
bash verif/regress/dv-riscv-arch-test.sh`
It seem like to used verilator& ISS to run this simulation.
However, I tried to use these two simulation types separately. (not at the same round of simulation)
Then, it works.

So, I think this problem just like what #1895 @valentinThomazic said, the test is timeout.
When using two simulation types, ex. verilator& ISS, at the same time, the simulation time will too long
that the error comes out or the simulation stop.
For another solution, I think the simulation time maybe can be modified more longer.

michael116688, 2024-03-19:
> When we face this kind of problem, we compare the log trace coming from Spike and RTL to understand the discrepancy, and try to solve the issue. Traces are under out_xxxx/

I tried to use the two simulation types (veri-testharness& Spike) separately. (not at the same round of simulation)

I use Spike first. It can completely run riscv-arch-test.
Than, when I use veri-testharness, it still have the same problem come out.
Is it still the same method to solve the issue?

valentinThomazic, 2024-03-26:
Your result is normal, the issue is coming from the RTL. Spike can run the test fine.

valentinThomazic, 2024-03-29:
Maybe this issue can be closed since you identified the cause and opened another issue about it.

michael116688, 2024-04-02:
OK

closed 2024-04-02 by michael116688