openhwfoundation/cva6 #1919: Getting fatal error while running riscv-arch-test from smoke-test.sh script

dvusingh, 2024-03-12:
Hi,

Today, I cloned the updated cva6_repo and ran the smoke-test.sh script.

python3 cva6.py --testlist=../tests/testlist_riscv-arch-test-cv64a6_imafdc_sv39.yaml --test rv64i_m-add-01 --iss_yaml cva6.yaml --target cv64a6_imafdc_sv39 --iss=$DV_SIMULATORS $DV_OPTS  --linker=../tests/riscv-arch-test/riscv-target/spike/link.ld

For riscv-arch-test I am getting the below error:

Tue, 12 Mar 2024 03:47:53 INFO     Found matched tests: rv64i_m-add-01, iterations:1
Tue, 12 Mar 2024 03:47:53 INFO     CVA6 Configuration is 
Tue, 12 Mar 2024 03:47:53 INFO     Compiling assembly test: /home/dvusingh/cva6_updated/verif/tests/riscv-arch-test/riscv-test-suite/rv64i_m/I/src/add-01.S
/home/dvusingh/cva6_updated/verif/tests/riscv-arch-test/riscv-test-suite/rv64i_m/I/src/add-01.S:18:10: fatal error: model_test.h: No such file or directory
   18 | **#include "model_test.h"**
      |          ^~~~~~~~~~~~~~
compilation terminated.

Is there any file missing in my repo?

Thanks

valentinThomazic, 2024-03-12:
Did you try redoing the 'quick setup' and 'running standalone simulations' sections of the readme ?

dvusingh, 2024-03-13:
I have done the setup given  in "quick_setup" guide.

I found some files are missing from the verif/tests/riscv-arch-test directory. When I added manually, the  riscv-arch-test is passing.

Thanks

valentinThomazic, 2024-03-13:
Did you ran the smoke test script beforehand ? It installs every test suites

dvusingh, 2024-03-13:
The issue is with riscv-arch-test installation in my system.

Now the issue is resolved

closed 2024-03-13 by dvusingh