openhwfoundation/cva6 #1937: 5 failed in smoke test

Tanishqgithub, 2024-03-16:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Respected Sir 

I am a new CVA-6 User and while running the smoke test using these command :
export DV_SIMULATORS=veri-testharness,spike
bash verif/regress/smoke-tests.sh

i got 5 fail can u please help me in this matter :
the log file are given below :
1.  iss_regr.log
[iss_regr_log.txt](https://github.com/openhwgroup/cva6/files/14623604/iss_regr_log.txt)

2.  add.log.iss
[add_log_iss.txt](https://github.com/openhwgroup/cva6/files/14623617/add_log_iss.txt)

3.  add-01.log.iss
[add-01_log_iss.txt](https://github.com/openhwgroup/cva6/files/14623619/add-01_log_iss.txt)

4.  cadd_01.log.iss
[cadd-01_log_iss.txt](https://github.com/openhwgroup/cva6/files/14623620/cadd-01_log_iss.txt)

5.  custom_test_template.log.iss
[custom_test_template_log_iss.txt](https://github.com/openhwgroup/cva6/files/14623622/custom_test_template_log_iss.txt)

6.  hello_world.log.iss
[hello_world_log_iss.txt](https://github.com/openhwgroup/cva6/files/14623623/hello_world_log_iss.txt)

7.  I-ADD-01.log.iss
[I-ADD-01_log_iss.txt](https://github.com/openhwgroup/cva6/files/14623624/I-ADD-01_log_iss.txt)

Vinay0203, 2024-03-17:
I am encountering the same issues. Can someone help to resolve them.

JeanRochCoulon, 2024-03-18:
These tests work on CI side and in local. You should have an env issue. Could you run them from scratch, that is to say cloning the cva6 repo from a new terminal window?

valentinThomazic, 2024-03-18:
Hi @Tanishqgithub, the name of your logs tells me that your version of the repo is a bit old : they should now have the config name before the test name.
I suggest you either pull master, update the submodules and then relaunch the tests or starting from scratch as @JeanRochCoulon suggested.

valentinThomazic, 2024-03-25:
Hey @Tanishqgithub and @Vinay0203, did you manage to solve the issue ?

JeanRochCoulon, 2024-04-29:
As no feedback from @Tanishqgithub for a while, we can suppose the github issue is solved. I close it.

closed 2024-04-29 by JeanRochCoulon

Tanishqgithub, 2024-04-29:
Respected Sir 
Sorry for the late response the issue was solved using your guidance 
Thank You