openhwfoundation/cva6 #1944: [TASK] Fix regression tests

cathales, 2024-03-19:
### Is there an existing CVA6 task for this?

- [X] I have searched the existing task issues

### Task Description

Make dashboard green for `master` branch.

### Required Changes

Fix all crashing and failing tests. They are in the `compliance` and `riscv-tests-p` jobs.

### Current Status

List of crashing tests on c430c6c34be0e24175e03d3c91478e349bfaaf4e (2024-01-05):

- In `compliance` (`verif/tests/testlist_riscv-compliance-cv32a6_embedded.yaml`)
    - `rv32uc-rvc`
    - `rv32ui-jal`
    - `rv32ui-bge`
    - `rv32ui-blt`
    - `rv32ui-bgeu`
    - `rv32ui-sw`
    - `rv32ui-lbu`
    - `rv32ui-sb`
    - `rv32ui-slti` Note: this test, compiling `slti.S`, is actually named `rv32ui-sw` in the test list. Looks like a typo.
    - `rv32ui-sra`
    - `rv32ui-srl`
    - `rv32ui-sh`
    - `rv32ui-lw`
    - `rv32ui-andi`
    - `rv32ui-srli`
    - `rv32ui-slli`
    - `rv32ui-beq`
    - `rv32ui-sll`
    - `rv32ui-addi`
    - `rv32ui-lh`
    - `rv32ui-and`
    - `rv32ui-xori`
    - `rv32ui-sub`
    - `rv32ui-slt`
    - `rv32ui-lb`
    - `rv32ui-or`
    - `rv32ui-lui`
    - `rv32ui-ori`
    - `rv32ui-bltu`
    - `rv32ui-fence_i`
    - `rv32ui-auipc`
    - `rv32ui-srai`
    - `rv32ui-jalr`
    - `rv32ui-xor`
    - `rv32ui-simple`
    - `rv32ui-lhu`
    - `rv32ui-bne`
    - `rv32ui-add`
    - `rv32ui-sltiu`
    - `rv32ui-sltu`
    - `rv32si-sbreak`
    - `rv32si-scall`
    - `rv32si-ma_fetch`
    - `rv32si-wfi`
    - `rv32si-csr`
    - Note that there are 2 failing tests, `rv32mi-scall` and `rv32mi-csr`, but at least the job runs to completion after removing the above tests.
- In`riscv-tests-p` (`verif/tests/testlist_riscv-tests-cv32a6_embedded-p.yaml`)
    - `rv32mi-p-scall`
    - `rv32si-p-csr`
    - `rv32si-p-ma_fetch`
    - `rv32si-p-scall`
    - `rv32si-p-wfi`
    - `rv32si-p-sbreak`
    - Note that there is 1 failing test, `rv32mi-p-csr`, but at least the job runs to completion after removing the above tests.

The crashes seem related to the two letters after `rv32`.

### Description of Done

Dashboard is green for `master` branch.

closed 2024-04-24 by github-actions[bot]

xiaoweish, 2024-05-09:
As to this case `rv32mi-csr`, this PR might help. https://github.com/openhwgroup/core-v-verif/pull/2424
[B Standard Extension for Bit Manipulation Instructions](https://drive.google.com/file/d/1SgLoasaBjs5WboQMaU3wpHkjUwV71UZn/view?usp=drive_link) is ratified on April 2024, listed in [Ratified Extensions](https://wiki.riscv.org/display/HOME/Ratified+Extensions)

JeanRochCoulon, 2024-05-30:
@xiaoweish  Thank for your help on this issue.
For information, the GCC version is 13.1 (aligned on CVA6 README)

valentinThomazic, 2024-07-05:
The embedded configuration is not tested anymore in gitlab ci, maybe this issue needs to be updated ?