openhwfoundation/cva6 #1945: Log Failure Despite Passing Tests: AXI4_RESET Violation in CVA6 Repository

akalboussi, 2024-03-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

### Description
I have executed a series of commands to run tests on the CVA6 repository. Despite the tests passing successfully, upon inspecting the logs (`Verif/sim/output_XX/vcs-gate-sim/lw.cv32a60x.log.iss`), I discovered that the tests have actually failed.

### Steps to Reproduce
```sh
git clone https://github.com/openhwgroup/cva6
cd cva6
git submodule update --init --recursive 
bash verif/regress/smoke_tests.sh 
source verif/sim/setup-env.sh
make -C pd/synth cva6_synth TARGET=cva32a60x
cd verif/sim
python3 cva6.py --testlist=../tests/testlist_riscv-tests-cv32a60x-p.yaml --test rv32ui-p-lw --iss_yaml cva6.yaml --target cv32a60x --iss=spike,vcs-gate $DV_OPTS

```

### Issue
Upon examining the logs, the following summary indicates that the simulation failed:
```sh
uvmt_cva6_tb.end_of_test: *** Test Summary ***

    FFFFFFFF   AAAAAA   IIIIII  LL        EEEEEEEE  DDDDDDD       
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FFFFF     AAAAAAAA    II    LL        EEEEE     DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA    II    LL        EE        DD    DD      
    FF        AA    AA  IIIIII  LLLLLLLL  EEEEEEEE  DDDDDDD       
    ----------------------------------------------------------
                       SIMULATION FAILED                    
                 test exit code = 0 (0x00000000)
    ----------------------------------------------------------
``` 
Further investigation into the error reveals: 

```sh
** Report counts by severity
UVM_INFO :   26
UVM_WARNING :    4
UVM_ERROR :    1
UVM_FATAL :    0
``` 
And specifically, the cause of the error is reported as:
```sh
UVM_ERROR @ 3.500 ns : uvma_axi_assert.sv(25) reporter [AXI4 protocol checks assertion] Violation of AXI4_RESET
```

Despite the simulations appearing to pass, the logs reveal errors indicating a failure, particularly due to an AXI4_RESET violation in the CVA6 repository. If anyone has insights into the root cause of this discrepancy or suggestions for resolution, your input would be greatly appreciated.

Gchauvon, 2024-03-19:
Reset is asserted before before 1st clock cycle leading to weird transition on the AXI agent side.

JeanRochCoulon, 2025-01-02:
@AEzzejjari is this issue fixed ?