openhwfoundation/cva6 #1946: [BUG] Passive mode of AXI agents does not work anymore

Gchauvon, 2024-03-19:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When using cva6.py with `--axi_active="no"` and vcs-uvm tesbench simulation will fail or crash.

JeanRochCoulon, 2024-05-19:
@AEzzejjari Can you status on this Github issue ?

AEzzejjari, 2024-05-20:
There is a problem with the coverage in passive mode. Once we merge the new agent, the problem will be fixed.

dvusingh, 2024-06-11:
Hi @AEzzejjari ,

I am facing the coverage related issue in case of passive mode, after taking the new agent changes.
I am getting the below mentioned uvm_info when test starts and then timeout occurs.

UVM_INFO @ 105.500 ns : uvmt_cva6_firmware_test.sv(159) uvm_test_top [TEST] Started RUN
UVM_INFO @ 783.500 ns : uvma_axi_covg.sv(265) uvm_test_top.env.axi_agent.axi_covg [uvma_axi_covg_c] READ ADDRESS ITEM DETECTED
UVM_INFO @ 783.500 ns : uvme_axi_covg.sv(279) uvm_test_top.env.cov_model.axi_covg [uvme_axi_covg_c] READ ADDRESS ITEM DETECTED
UVM_INFO @ 783.500 ns : uvme_axi_ext_covg.sv(272) uvm_test_top.env.cov_model.axi_ext_covg [uvme_axi_ext_covg_c] READ REQ ITEM DETECTED
UVM_INFO @ 789.500 ns : uvme_axi_covg.sv(285) uvm_test_top.env.cov_model.axi_covg [uvme_axi_covg_c] READ DATA ITEM DETECTED

Also, when I am disabling the coverage ( cov_model_enabled=0) timeout error is coming.

Thanks