openhwfoundation/cva6 #1948: Running CVA6 in Firesim+Chipyard

hakase56557, 2024-03-21:
Hello,
I'm trying to run CVA6 with Firesim on an F1 instance, but sadly no luck so far. I'm able to build an AGFI, but no luck with booting Linux (nor with baremetal stuff). 
I am successfully able to run an older CVA6 version (from 2021) that is bundled with Chipyard, but no luck with a newer CVA6 release (from 2023). I was wondering if there were any changes in particular between the old and the newer versions which I need to keep in mind?

 I was hoping if anyone who has managed to run it may be able to help. It would be greatly appreciated.

Thanks!

JeanRochCoulon, 2024-03-21:
How do you succeed to use Firesim on recent CVA6 version ? It has been removed some time ago.

hakase56557, 2024-03-22:
Oh I'm succeeding to run an older [version](https://github.com/openhwgroup/cva6/tree/6000e32b4181304901adc4c8d72870ebc94473a5), not a recent one.

> It has been removed some time ago.

Are there changes because of which it's not possible to run on Firesim? (Apologies if it's a beginner question, as I'm relatively new to RTL stuff)

JeanRochCoulon, 2024-03-22:
Firesim cannot work on new version of cva6. If you think we answer the question, can you close the Github Issue ?

closed 2024-03-25 by JeanRochCoulon