openhwfoundation/cva6 #1953: [BUG] Illegal instruction not raising exception

AyoubJalali, 2024-03-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
After merging the [H extension](https://github.com/openhwgroup/cva6/commit/86e1408666eb62b5380f31edfb4354d887a81087), I run a regression to see if, the added code not affecting our code coverage on not supported H extension target.
What I saw after a regression, a lot of falling tests due to mismatch between the CVA6, and Spike, after reproducing the failing tests, I had some illegal instruction not raising exception on the CVA6 side.
as an example : 0xfeb00073 also 0x88adcb73 and more.

It seems as a SYSTEM instruction, but I don't know if it's H extension instructions to be honest, what I'm sure about the this BUG, appear after the merge of the H extension, and with a target that doesn't support H extension (RVH = 0).

Spike Version: 1.1.1-dev 958a5cb6
CVA6 HASH: 86e1408666eb62b5380f31edfb4354d887a81087

So to reproduce the BUG here is 2 examples of assembly code has the BUG : 

[BUG1.txt](https://github.com/openhwgroup/cva6/files/14724610/BUG1.txt)
[BUG2.txt](https://github.com/openhwgroup/cva6/files/14724658/BUG2.txt)

AyoubJalali, 2024-03-22:
@ninolomata I couldn't assign it to you, so I leave this comment to see the issue.

ninolomata, 2024-03-22:
@AyoubJalali see PR #1955 
Thank you!

JeanRochCoulon, 2024-03-25:
@AyoubJalali Can you check that the issue is fixed ?   Then close it

closed 2024-03-25 by AyoubJalali