openhwfoundation/cva6 #1964: [BUG] rv64ui-v test log overwritten by rv64ui-p

AngelaGonzalezMarino, 2024-03-26:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

In the smoke tests, these two add tests are executed sequentially:
https://github.com/openhwgroup/cva6/blob/f73724b7e2cc6752d543b8ddaf61f66a9eb50042/verif/regress/smoke-tests.sh#L43C1-L44C182

Both of them store the result in a file with name add.cv64a6_imafdc_sv39.log, so the second one overwrites the first.

This is why sometimes the CI reports 17/18 tests pass, with the first add test failing (rv64ui-v), but the log inside says PASS (because it is the log of the second one)

valentinThomazic, 2024-04-03:
Hey @AngelaGonzalezMarino, the problem should now be resolved !

AngelaGonzalezMarino, 2024-04-03:
perfect, thanks!

closed 2024-04-03 by AngelaGonzalezMarino