openhwfoundation/cva6 #1965: [BUG] Incorrect Update of ‘minstret’ Register on CSR Operations

youzi27, 2024-03-27:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

**Describe the bug:**
In the CVA6 implementation, executing CSR operations like` csrrw s2, minstret, s8` erroneously updates the `minstret` register. Before execution, if minstret holds a value `x`, the instruction should ideally write `x` into `s2`. 
However, it mistakenly updates minstret to `x+1`, incorrectly incrementing the instruction count.

**Expected behavior:**
The correct behavior should not alter the `minstret` register's value upon reading with CSR operations. The minstret should accurately reflect the total number of instructions executed since reset, without being affected by CSR read operations.

**To Reproduce:**
1. Note the current value of `minstret` (`x`).
2. Execute a CSR operation like `csrrw s2, minstret, s8`.
3. Observe the value written to `s2` and the new value of minstret.
4. The operation incorrectly increments minstret to `x+1` instead of maintaining its value.

youzi27, 2024-03-27:
log info: 
```
core   0: 0x0000000080002016 (0xb02024f3) csrrs   s1, minstret, zero
3 0x0000000080002016 (0xb02024f3) x 9 0x00000000000008fc
core   0: 0x000000008000201a (0x7b173c5b) unknown
3 0x000000008000201a (0x7b173c5b) x24 0x0000000000000001
core   0: 0x000000008000201e (0xb0202973) csrrs   s2, minstret, zero
3 0x000000008000201e (0xb0202973) x18 0x00000000000008fe
```

JeanRochCoulon, 2024-03-27:
Thanks for the bug detection. Feel free to submit a PR if you like.

youzi27, 2024-03-28:
Hi @JeanRochCoulon !

I checked the code, and before incrementing the `minstret` register, it checks whether an exception instruction has been generated. 

See: https://github.com/ninolomata/cva6/blob/62bdf1159498c85a8dfbff4399ea26bc3a011cc9/core/csr_regfile.sv#L809-L811

Therefore, the root cause of this problem should be an issue with instruction decoding, leading to the unintended side effect of `minstret` being incorrectly incremented.

I am currently still testing, and I will try to fix it when I have time in the future.

Thank you.

JeanRochCoulon, 2025-01-02:
As MINSTRET (will) not be supported by CV32A65X, add the notCV32A65X flag