openhwfoundation/cva6 #1966: [BUG]  Unexpected Behavior in CVA6 with Illegal Instruction Altering Register Values

youzi27, 2024-03-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

**Describe the bug:**
In the CVA6 RISC-V core implementation, an illegal instruction (`0x49254b5b`) causes an unexpected modification to register values(`x22|s6`,register ). This issue deviates from the expected behavior where illegal instructions should trigger an exception without causing unintended side effects to the system's state, particularly not altering the contents of registers not explicitly targeted by the instruction.

**Expected behavior:**
Upon encountering an illegal instruction, the CVA6 core should raise an illegal instruction exception and leave the state of all registers unchanged except for those directly involved in the exception handling process (e.g., updating the mepc and mcause registers). The core **should not** modify the value of any other registers as a side effect of encountering an illegal instruction.

**To Reproduce:**

1. Write a simple RISC-V assembly program that includes the illegal instruction `0x49254b5b` at a known point in the execution flow.
2. Execute the program on the CVA6 core, ensuring that the execution context before the illegal instruction is well-documented, especially the state of all registers.
3. Observe the behavior of the core upon reaching the illegal instruction, with a particular focus on the state of registers before and after the instruction is encountered.
4. Verify that an illegal instruction exception is raised, and compare the register states to confirm any unintended modifications.

**The log is as follows**
```
core   0: 0x0000000080002018 (0x49254b5b) unknown
3 0x0000000080002018 (0x49254b5b) x22 0x000000000000000a
core   0: 0x000000008000201c (0x00000001) c.nop
3 0x000000008000201c (0x0001)
core   0: 0x000000008000201e (0x49254b5b) unknown
3 0x000000008000201e (0x49254b5b) x22 0x000000000000000a
```

youzi27, 2024-03-28:
Upon testing, it has been found that there are many such instructions, each altering different registers.

youzi27, 2024-03-28:
This issue has two parts: first, an illegal instruction doesn't cause an expected exception(issue #1967 and #1953); second, it wrongly changes a register's value(this issue).

JeanRochCoulon, 2025-01-02:
@AyoubJalali   If this issue is true, you should have detected it?  What do you think ?

AyoubJalali, 2025-01-02:
For the CV32A65X the RTL does the right thing :
 
1- raise exception for the unknown instruction 
2- doesn't change x22 | s6 register state

[image]

N.B : 
My explication is that you're using a target that support the mentioned instruction, try to decode it by hand and compare it with the supported instructions of you're RTL target, don't trust the tracer (if it write "unknown" perhaps the tracer doesn't know the instruction name)

youzi27, 2025-01-03:
Thank you for your response regarding this issue. However, quite a bit of time has passed (almost ten months), so perhaps you could consider testing with the older version (around commit: f73724b). From the screenshot you provided, it does seem that the illegal instruction exception is now being triggered correctly. But given the long timeline, pinpointing which commit actually fixed this bug is quite difficult. If you're confident that this issue has been resolved in the latest version, feel free to close this issue at any time.

Just a note: Regarding the `0x49254b5b` instruction, I decoded it using SPIKE and manually checked that its opcode doesn't correspond to any operation in the published specifications. However, the previous CVA6 did execute it and modified register values.

If you have any questions or would like to discuss further, feel free to reach out!

JeanRochCoulon, 2025-01-03:
Thanks @AyoubJalali and @youzi27 for your investigations. I think we can close it.

closed 2025-01-03 by JeanRochCoulon