openhwfoundation/cva6 #1967: [BUG] Fails to Trigger Illegal Instruction Exception for some illegal Instructions

youzi27, 2024-03-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

**Describe the bug:**
The CVA6 does not trigger an illegal instruction exception when encountering the illegal instruction like `'0x5162e55b'`. This issue is reminiscent of the previously reported issue #1953 . Despite switching to a newer version of the core (commit: f73724b7e2cc6752d543b8ddaf61f66a9eb50042), the problem persists for certain instructions, including `‘0x5162e55b’`. In addition, there are the instructions mentioned in issue #.

**Expected behavior:**
Upon encountering an illegal instruction, such as ‘0x5162e55b’, the CVA6 core is expected to trigger an illegal instruction exception. 

**The log is as follows**
```
core   0: 0x0000000080002020 (0x5162e55b) unknown
```

JeanRochCoulon, 2024-03-28:
Thnak you for this feedback. If I understand this bug was already detected and reported in #1953 and #1966, why to create a new GitHub issue. Do I misunderstand ?

JeanRochCoulon, 2024-03-28:
I like getting feedbacks on CVA6 implementation, so let's go on with your work and contributions.
When a github issue is open, it means that it is not fixed. When someone fixes it, we close the Github issue.
That is why I would ask you to comment the existing Github issues (is needed), and close the current one. Thanks !

youzi27, 2024-03-28:
Thank you for your encouraging response. I appreciate the clarity regarding the Github issue management process and will follow your guidance by commenting on the existing issues as needed and closing the current one. Thanks again for your support!😊

closed 2024-03-28 by youzi27