openhwfoundation/cva6 #1972: [BUG] Spike crush when get multiple extension

AyoubJalali, 2024-03-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, 
So this issue related to Spike, when it get multiple extension (cvxif & cv32a60x) spike crush end we get the following error : 
[image]

This issue is detected with the tandem mode, also the trace and compare mode
Spike version : `f368b1f40`

zchamski, 2024-03-29:
Strictly speaking, this is a core-v-verif issue and belongs there.  The root cause is an incorrect coding in Spike CV-X-IF extension, causing Spike to abort upon decoding a CV-X-IF instruction if at least one more instruction is loaded.

The incorrect behavior is reproducible by manually invoking `spike --extension=cvxif --extension=cv32a60x` on a test involving CV-X-IF instructions, e.g., any of the tests generated when running `bash verif/regress/dv-generated-xif-tests.sh`.

The fix is already devised and tested; the PR will be submitted shortly.

AyoubJalali, 2024-03-29:
Thanks for the update !!

zchamski, 2024-04-02:
This issue is fixed in CVA6 master commit openhwgroup/cva6@4f9dd2af (bump of core-v-verif to include the merge of openhwgroup/core-v-verif#2401).

AyoubJalali, 2024-04-02:
I'll double check the fix, before I close this issue !!
I'll assign this issue to me for now

AyoubJalali, 2024-04-02:
> This issue is fixed in CVA6 master commit [4f9dd2af](https://github.com/openhwgroup/cva6/commit/4f9dd2af) (bump of core-v-verif to include the merge of [openhwgroup/core-v-verif#2401](https://github.com/openhwgroup/core-v-verif/pull/2401)).

After checking !! the related PR fix the Bug

AyoubJalali, 2024-04-02:
thanks @zchamski !!