openhwfoundation/cva6 #1973: [BUG] Spike Dons't set bitmanip bit in the MISA

AyoubJalali, 2024-03-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
Hello, Spike has that following reset value on MISA : `0x40001104`, and decoding that value using the Table 3.2: Encoding of Extensions field in misa in privlied RISCV spec (page 31)
image

you can see clearly that bitmanip extension is decoded in the second bit, so the reset value isn't correct, normally as the bitmanip extension is enabled, the reset value should be : `0x40001106`

zchamski, 2024-04-17:
The normative definition of the `B` bit has changed at least twice in the recent months:
* in Privileged ISA spec (v1.12 dated 20211203, ratified) it was *Tentatively reserved for Bit-Manipulation extension*;
* in subsequent ASCIIdoc releases of the Privileged ISA spec prior to commit 9f9130e4 dated 2024-03-25 it was marked as *Reserved* (meaning it should be WARL, read-only zero);
* since commit 9f9130e4 it is labeled as *B extension* with the remark that:

> *When "B" bit is 1, the implementation supports the instructions provided by the **Zba**, **Zbb**, and **Zbs** extensions. When "B" bit is 0, it indicates that the implementation may not support one or more of the **Zba**, **Zbb**, or **Zbs** extensions.*

This means two things:
* Spike is likely to be updated any day now (if it did not happen already)
* We will need to upgrade Spike and review all open Spike/tandem issues again :grin:

zchamski, 2024-04-17:
Well, Spike has already been updated wrt. the new definition of the `B` bit (https://github.com/riscv-software-src/riscv-isa-sim/commit/0bf0a60ad870ba3d59f87c03b41b6d06ca5fa30c) => an update of our vendorized Spike tree becomes critical.

MarioOpenHWGroup, 2024-04-17:
First we need to have an equivalent version of the base + patches that I think currently is broken :cry:

zchamski, 2024-04-18:
> First we need to have an equivalent version of the base + patches that I think currently is broken 😢

Yes, prior to a vendorized upgrade we need to clean up our own act 😛

zchamski, 2024-04-18:
> In my PR I have added a param to toggle the B bit as a param too

We should be careful about possible inconsistencies this could introduce (such as "B bit cleared but Zba+Zbb+Zbs enabled").  I'm strongly in favor of structural solutions where possible, based on upstream updates and/or root cause fixes to be upstreamed (see issue #1974).

MarioOpenHWGroup, 2024-06-11:
Solved

closed 2024-06-11 by MarioOpenHWGroup