openhwfoundation/cva6 #1974: [BUG] Spike enabling `MTIP` by default

AyoubJalali, 2024-03-28:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello,
After running a rest CSR test, I observe that reset value read from MIP was : `0x00000080`, an according to [CSR Spec](https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/CSR_CV32A60X.html), the MIP reset value should be : `0x0`

zchamski, 2024-04-04:
This Spike behavior seems to be an artefact of the way the CLINT model is initialized: when its constructor is called, for each hart it triggers a comparison of `mtime` against `mtimecmp + 0`. Since the compare relation used is `>=` and both registers have value 0, the comparison yields `true` and the interrupt is raised (see file `riscv/clint.cc` in Spike).

Fixing this requires further analysis of Spike's CLINT implementation because of interactions with the `realtime` mtime-tracking mode of Spike.

AyoubJalali, 2024-04-04:
> This Spike behavior seems to be an artefact of the way the CLINT model is initialized: when its constructor is called, for each hart it triggers a comparison of `mtime` against `mtimecmp + 0`. Since [...]

thanks for the update @zchamski

zchamski, 2024-04-22:
After a deeper analysis of code traces, here's what happens:

- by default Spike enables the CLINT controller;
- at every instruction commit, the simulation engine advances the timer by 32 (!);
- the CLINT detects the advance of the timer and raises the MTIP bit after one instruction:
```
[SPIKE] Proc 0 | ISA: rv32imc_zba_zbb_zbs_zbc_zicsr_zifencei PRIV: m
[SPIKE] Initializing memories...
[SPIKE] MIP register prior to current step: 0x0
core   0: 0x00010000 (0x00100413) li      s0, 1
core   0: 3 0x00010000 (0x00100413) x8  0x00000001
[SPIKE] Issuing clint->increment(32)
[SPIKE] MIP register prior to current step: 0x80
```
We are here in the middle of interrupt territory, with a volatile register  (`MTIME`) as a bonus... To achieve precise interrupt injection in such a context we should take a global and systematic approach - adding hooks rather than hacks.

My proposal:

- if this is acceptable on the RTL side, disable CLINT unless explicitly required;
- use value injection from RTL to control the value of `MTIME` and raise any events associated with the corresponding changes.

MarioOpenHWGroup, 2024-06-11:
Solved

closed 2024-06-11 by MarioOpenHWGroup