openhwfoundation/cva6 #1975: [BUG] ’zext.h‘ is considered an illegal instruction.

youzi27, 2024-03-29:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

After enabling extensions `rv64gc_zba_zbb_zbs_zbc_zicsr_zifencei`, CVA6 cannot execute the '`zext.h`' instruction correctly, resulting in an illegal instruction exception. This seems to indirectly explain the issues with issue #1758 and issue #1918.

**The log is as follows**
`ILLEGAL_INSTR exception @ 0x00000000800041d6`

JeanRochCoulon, 2024-04-11:
#2032  should fix the bug. Can you verify and close if agree. Thanks

youzi27, 2024-04-11:
After testing, `zext.h` has successfully run following the fix. The log is as follows:
```
core   0: 0x000000008000403c (0x0803c1bb) zext.h  gp, t2
1 0x000000008000403c (0x0803c1bb) x 3 0x0000000000000000
```

I will close this issue. Thank you very much!

closed 2024-04-11 by youzi27