openhwfoundation/cva6 #1981: Trap-return instructions set MPRV of mstatus wrongly

salaheddinhetalani, 2024-03-31:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Trap-return instructions, `MRET` and `SRET`, set `MPRV` of `mstatus` or `sstatus` wrongly.

### RISC-V Specification
Unprivileged ISA: [20191213](https://drive.google.com/file/d/1s0lZxUZaa7eV_O0_WsZzaurFLLww7ou5/view) | Privileged Architecture: [20211203](https://drive.google.com/file/d/1EMip5dZlnypTk7pt4WWUKmtjUKTOkBqh/view) | External Debug Support: [0.13.02](https://drive.google.com/file/d/1XSPqW3WIfflIMeuGJBxGTMOEyeuRAQ5F/view)
> To support nested traps, each privilege mode x that can respond to interrupts has a two-level
stack of interrupt-enable bits and privilege modes. xPIE holds the value of the interrupt-enable bit
active prior to the trap, and xPP holds the previous privilege mode. The xPP fields can only hold
privilege modes up to x, so MPP is two bits wide and SPP is one bit wide. When a trap is taken
from privilege mode y into privilege mode x, xPIE is set to the value of x IE; xIE is set to 0; and
xPP is set to y.

> An MRET or SRET instruction is used to return from a trap in M-mode or S-mode respectively.
When executing an xRET instruction, supposing xPP holds the value y, xIE is set to xPIE; the
privilege mode is changed to y; xPIE is set to 1; and xPP is set to the least-privileged supported
mode (U if U-mode is implemented, else M). If xPP̸=M, xRET also sets MPRV=0.

> An MRET or SRET instruction that changes the privilege mode to a mode less privileged than M
also sets MPRV=0.

### Example Scenario

As shown below, the following sequence of instructions happens: 
> CSRRS x1, mstatus, x1 -> Unknown -> SRET

The instruction `SRET` is decoded at `t_id` and executed in machine mode resulting in returning from a trap and updating the architecture state at `t_arch_update`. As a result, the privilege level is set to user mode, since `mstatus.SPP` is unset, and all `mstatus` fields are updated correctly except `MPRV`, where it's not unset.

[image]

### Steps to Reproduce

Git Hash: [de2e254c](https://github.com/openhwgroup/cva6/commit/de2e254cd4c2d3e3a6d4def1652918e8f2535512) | TARGET_CFG: [cv32a6_imac_sv0](https://github.com/openhwgroup/cva6/blob/master/core/include/cv32a6_imac_sv0_config_pkg.sv) | VCD: [issue_4.vcd](https://github.com/openhwgroup/cva6/files/14817337/issue_4.zip)

### Component
**Component:RTL**

---
Product: Questa OneSpin Solutions
App: Questa Processor App
Tool's version: 2024.1_1

JeanRochCoulon, 2025-01-02:
As Atomic extensino is not supported by CV32A65x, add notCV32A65X flag

bobsystone-cmd, 2026-09-06:
A complementary RV64 non-H manifestation of this xRET behavior is present in current CVA6 RTL:

 - Commit: 6cb200105fb9441d170e45786125a737fab98e91
 - Config: cv64a6_imafdc_sv39 (RVH=0)
 - Path: MRET from M-mode to S-mode

 Unlike the SRET-to-U scenario described above, this test sets mstatus.MPRV=1 and MPP=S, executes MRET, then traps from S-mode back to M-mode and checks mstatus. MPRV remains set; the self- checking Verilator test exits with tohost=6 after 35,780 cycles.

 At VCD time #12351, mret=1 and mstatus_d=0x8000000a00026880, so MPRV bit 17 remains set. The same value is registered into mstatus_q at #12353. The ELF, VCD, and source are attached.

[image]

 A minimal fix appears to be moving the MPRV clears outside the CVA6Cfg.RVH guards in csr_regfile.sv, while retaining only virtualization-specific state updates inside those guards:

 // MRET
 if (mstatus_q.mpp != riscv::PRIV_LVL_M)
   mstatus_d.mprv = 1'b0;

 if (CVA6Cfg.RVH) begin
   // existing virtualization-specific updates
 end

 // SRET legal path
 mstatus_d.mprv = 1'b0;

 if (CVA6Cfg.RVH) begin
   // existing virtualization-specific updates
 end

 Is a resolution or fix already in progress for this issue?