openhwfoundation/cva6 #1997: [Possible Issue] Instruction retire counters are incremented and updated simultaneously

salaheddinhetalani, 2024-04-02:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Instruction retire counters `minstret` and `minstreth` are incremented and updated (written by software) at the same time. Is this intended?

### RISC-V Specification
Unprivileged ISA: [20191213](https://drive.google.com/file/d/1s0lZxUZaa7eV_O0_WsZzaurFLLww7ou5/view) | Privileged Architecture: [20211203](https://drive.google.com/file/d/1EMip5dZlnypTk7pt4WWUKmtjUKTOkBqh/view) | External Debug Support: [0.13.02](https://drive.google.com/file/d/1XSPqW3WIfflIMeuGJBxGTMOEyeuRAQ5F/view)
> The minstret CSR counts the number of instructions the hart has retired. The mcycle and minstret registers
have 64-bit precision on all RV32 and RV64 systems.

> The counter registers have an arbitrary value after the hart is reset, and can be written with a
given value. Any CSR write takes effect after the writing instruction has otherwise completed.

> When MXLEN=32, reads of the mcycle, minstret, and mhpmcountern CSRs return bits 31–0 of
the corresponding counter, and writes change only bits 31–0; reads of the mcycleh, minstreth,
and mhpmcounternh CSRs return bits 63–32 of the corresponding counter, and writes change only
bits 63–32.

### Example Scenario

As shown below, the following sequence of instructions happens: 
> CSRRW x9, minstret, x9 -> CSRRC x0, minstret, x2

The instruction `CSRRC x0, minstret, x2` is decoded at `t_id` and executed in machine mode resulting in updating the architecture state at `t_arch_update`. As a result, `minstret` is updated to have been incremented and then cleared at the same time.

[image]

### Steps to Reproduce

Git Hash: [de2e254c](https://github.com/openhwgroup/cva6/commit/de2e254cd4c2d3e3a6d4def1652918e8f2535512) | TARGET_CFG: [cv32a6_imac_sv0](https://github.com/openhwgroup/cva6/blob/master/core/include/cv32a6_imac_sv0_config_pkg.sv) | VCD: [issue_17.zip](https://github.com/openhwgroup/cva6/files/14836576/issue_17.zip)

### Component
**Component:RTL**

---
Product: Questa OneSpin Solutions
App: Questa Processor App
Tool's version: 2024.1_1

JeanRochCoulon, 2024-04-03:
Thanks @salaheddinhetalani for this study (all the bug issues opened these last days). The bug are very well linked to the specification. I am sure it will help the understanding.
BTW, Today the Questasim tool is not fully supported by cva6 repo. Is there an interest from you in providing scripts to simulate with such tools ?

salaheddinhetalani, 2024-04-05:
Hi @JeanRochCoulon, it's my pleasure. As for providing scripts to reproduce the issues reported using Questasim, I have already created SV testbenches based on the counterexamples of each issue. Would you be interested to have them shared here as well. Otherwise, the attached VCD files should be good enough to reproduce. Cheers!