openhwfoundation/cva6 #2010: an overlap with "ZEXTH" instruction

bfarnaghi, 2024-04-05:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I am developing a cryptography extension on this core. While I was testing added instructions, I found out that the "pack" instruction in the "Zbkb" extension overlaps with the "ZEXTH" instruction. I commented on line 780 of "decoder.sv" to manage this instruction by coprocessor.

ASintzoff, 2024-04-10:
Could you provide more information?

- Which version of CVA6 RTL are you referring to?
- Which Zbkb instruction(s) are overlapping?

bfarnaghi, 2024-04-10:
Instruction "ZEXTH" overlaps with the "pack" instruction in the scalar cryptography extension. 

This instruction is decoded in line 780 of the "decoder.sv" in the latest version which is 5.0.1. 
It also overlaps with the previous version but in line 601 of "decoder.sv".

ASintzoff, 2024-04-10:
Currently, CVA6 does not support Zbkb extension. `PACK` reuses the `ZEXT.H` encoding when `rs1` field (bit 20:24) is equal to 0. Therefore, if you would like to add `PACK` instruction, some modifications are needed in CVA6 RTL.

JeanRochCoulon, 2024-04-11:
#2032 should fix the bug. Can you verify and close if agree. Thanks

closed 2024-04-12 by bfarnaghi