openhwfoundation/cva6 #2012: [BUG] Unexpected Behavior in M Extension Instructions with Identical Operands

youzi27, 2024-04-08:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

When executing instructions from the M extension (such as `div a0, a0, a0`, `divu a0, a0, a0`, etc.) and `rd = rs1 = rs2`, the cva6 cannot operate normally to obtain the expected value.

ASintzoff, 2024-04-10:
Could you provide 
- the CVA6 version you are using (commit hash),
- the value of `a0` before and after the execution of `div a0, a0, a0`, `divu a0, a0, a0` and others if you have.

youzi27, 2024-04-11:
Hi @ASintzoff ,

I discovered that this issue originates from the **standalone simulations** provided by cva6. It results in an inability to exit normally before quitting when `a0=0b1` (which becomes `0b10` after left shifting). A PR has already been submitted.

closed 2024-04-11 by youzi27