openhwfoundation/cva6 #2026: [BUG] uvme_exception_covg and uvme_illegal_instr_covg not compliant LRM

CoralieAllioux, 2024-04-10:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

We are currently porting the verification to be compliant with Cadence Xcelium simulator. The objective is to use the current verification flow and adapt it, to add the Cadence simulator.
This issue concerns the _cva6/verif/env/uvme/cov/uvme_exception_covg.sv_ and _cva6/verif/env/uvme/cov/uvme_illegal_instr_covg.sv_.

For **uvme_exception_covg.sv**, the issue is coming from the following code:
`bins BREAKPOINT_EXC_RAISED = (NO_EXCEPTION => BREAKPOINT => NO_EXCEPTION);`
A bin name cannot be used inside a bin with a transition, xcelium does not allow it.

For **uvme_illegal_instr_covg.sv**, the issue is about the covergroup `cg_illegal_i`: in particular coverpoints **cp_illegal_funct3** and **cp_illegal_funct7**. For example:
`bins ILLEGAL_NOPCODE_FUNCT7[3] = {[0:$]} with (!(item inside {legal_i_funct7})) iff (!(instr.rvfi.insn[6:0] inside {legal_i_opcode}));`.
A bin defined that way (sequence of with / inside / iff / inside) is not compliant.

From now, we would like to get a solution to make work those covergroups for xcelium, which we still not manage to fix: we need a better understanding of the purpose of those lines.
We already discussed a bit with @AyoubJalali. Could you please better explain what you meant to achieve? In order to better look at potential solutions. Thank you!

AyoubJalali, 2024-07-02:
For the first issue, it's fix on master 
For the _legal_i_opcode_ my intend is to sample my bins only if we have the right opcode or the wrong opcode, the legal_i_opcode is define in https://github.com/openhwgroup/cva6/blob/0721ebb609044a4504be2d99751a0328856575bb/verif/env/uvme/uvme_cva6_pkg.sv#L63

OlivierBetschi, 2025-05-07:
For uvme_illegal_instr_covg.sv, I think the problem comes from legal_i_opcode being a queue. Only constants are allowed in covergroup expression.