openhwfoundation/cva6 #2029: ZCB RISCV compilor in smoke-gen_tests

CoralieAllioux, 2024-04-11:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I would have a question about the compilator used for script `verif/regress/smoke-gen_tests.sh`. It seems that it needs a custom one with ZCB instruction. This is actually one of the scripts we run for regression on the cva6 in our environment. But without that compilator we are not able to do it.

This is line 10 of the concerned script:
```
if [ -n "$RISCV_ZCB" ]; then
  echo "Using RISCV_ZCB to support Zcb extension"
  RISCV=$RISCV_ZCB
fi
```

Could we have more information about it? Could it be possible to have a more generic version we can reproduce?
Thank you for your time

JeanRochCoulon, 2024-04-11:
Ah, ok, I did not understand you were speaking about smoke-gen-test.sh (and not smoke-tests.sh). The GCC13.1 version we are using for the regression does not support Zcb extension. That is why we use another GCC supporting the Zcb for the smoke-gen-tests.sh. As this GCC version is not an official release, we do not switch to it.
As soon as an official compiler (GCC or LLVM) will be released, we will switch to it.

JeanRochCoulon, 2024-04-11:
If needed, we can provide you the link of the GCC supporting Zcb

CoralieAllioux, 2024-04-11:
I figure we can wait for the official release. We'll stick to `smoke-tests.sh` and `dv-csr-embedded-tests.sh` for now then. It should be sufficient to test our PR on vcs-uvm before publishing.
Thanks @JeanRochCoulon

CoralieAllioux, 2024-04-11:
Do I close it now? Or should it be closed when the GCC supporting Zcb is released and integrated in cva6?

JeanRochCoulon, 2024-04-11:
Yes, you can close it. Tha plan is to give the opportunity to switch to LLVM in June. This will be revisited et this time. No need of github issue.

closed 2024-04-11 by CoralieAllioux